limbo018/DREAMPlace #2: ImportError: No module named dreamplace

thesukantadey, 2019-06-26:
Please see this error, while I try to execute the following. This maybe a silly python error, but I am not able to fix it. I am using python 2.7. There is a file __init__.py inside the dreamplace directory, yet python is not able detect dreamplace as a module.

$ python dreamplace/Placer.py test/ispd2005/adaptec1.json
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 15, in <module>
    import PlaceDB
  File "/home/dey/Workspace/DREAMPlace/dreamplace/PlaceDB.py", line 14, in <module>
    import dreamplace 
ImportError: No module named dreamplace

limbo018, 2019-06-26:
Just to confirm. Did you run from the repo directory or the install directory?

limbo018, 2019-06-26:
> Please see this error, while I try to execute the following. This maybe a silly python error, but I am not able to fix it. I am using `python 2.7`. There is a file` __init__.py` inside the [...]

Just to confirm. Did you run from the repo directory or the install directory?

thesukantadey, 2019-06-26:
I executed it from the repo directory where I cloned the DREAMPlace project from git.

limbo018, 2019-06-26:
> I executed it from the repo directory where I cloned the DREAMPlace project from git.

You need to execute from the install directory. The repo directory is not for execution. It is just for building.

limbo018, 2019-06-26:
Please check commit [44dad80](https://github.com/limbo018/DREAMPlace/commit/44dad806dbda764be19856a445d85ca63986092a) or pull the latest version. Will assume the issue has been resolved if no response.

thesukantadey, 2019-06-27:
Sorry for the late reply.

Actually, I have not given any specific install directory location while building and installing dreamplace. It installed in `/usr/local` directory. From the `/usr/local` location also python is not able to detect `dreamplace` as a module. It must be some `python` error, will try to see.  Maybe I will try to reinstall it with `python 3`.

closed 2019-06-27 by thesukantadey

choltz95, 2019-07-08:
Did you resolve this issue? I'm having the same problem after I set up the environment in colab:
https://colab.research.google.com/drive/1hS42ZegFZhjO28On1Hk_1muC1opasBxA

limbo018, 2019-07-09:
> @thesukantadey Did you resolve this issue? I'm having the same problem after I set up the environment in colab:
> https://colab.research.google.com/drive/1hS42ZegFZhjO28On1Hk_1muC1opasBxA
> 
> Tried running dreamplace/Placer.py --help from both the repo directory and /usr/local

Hi choltz95, I got to reproduce the issue with Python 2.7. Now the latest commit should have fixed it (commit [d36f1a5](https://github.com/limbo018/DREAMPlace/commit/d36f1a59add497837f84ffbbb559ac63363eca1a)) This issue only shows up with Python 2, while Python 3 works fine.

jeshu54, 2020-05-27:
> Just to confirm. Did you run from the repo directory or the install directory?

Some how I am not able to run make or make install

I am using Mac.
cmake .. -DCMAKE_INSTALL_PREFIX=Documents/GitHub/Hubmaster/DREAMPlace/ I am getting a lot of errors when I run this code. 

after this when I run Make, below error is given

[image]
Can you please advice what I am doing wrong?

limbo018, 2020-05-29:
> > > Please see this error, while I try to execute the following. This maybe a silly python error, but I am not able to fix it. I am using `python 2.7`. There is a file` __init__.py` inside the `dreamplace` directory, yet python is not able detect `dreamplace` as a module.
> > > `$ python dreamplace/Placer.py test/ispd2005/adaptec1.json`
> > > `Traceback (most recent call last):`
> > > ` File "dreamplace/Placer.py", line 15, in <module>`
> > > ` import PlaceDB`
> > > ` File "/home/dey/Workspace/DREAMPlace/dreamplace/PlaceDB.py", line 14, in <module>`
> > > ` import dreamplace`
> > > `ImportError: No module named dreamplace`
> > 
> > 
> > Just to confirm. Did you run from the repo directory or the install directory?
> 
> @limbo018
> Some how I am not able to run make or make install
> 
> I am using Mac.
> cmake .. -DCMAKE_INSTALL_PREFIX=Documents/GitHub/Hubmaster/DREAMPlace/ I am getting a lot of errors when I run this code.
> 
> after this when I run Make, below error is given
> 
> [image]
> 
> Can you please advice what I am doing wrong?

Did the cmake step finish without any error? It looks like no Makefile generated under the build folder. 

To be honest, I do not recommend to build under MAC or Windows. They are quite different from linux.

jeshu54, 2020-05-30:
@limbo018  Make and make install ran with no errors, but install directory was not generated in root directory.