limbo018/DREAMPlace #4: Suboptimal CPU performance

gessfred, 2019-07-04:
Hi, would you happen to have any tips on how to properly install PyTorch(+other dependencies) in order to get the best performance out of a CPU-only environment. 
I could not get optimal performance when I ran it on a 8 core i7 CPU and the performance was only affected a little when I changed the `num_threads` parameter in the `.json` file (using System Monitor I was able to see that they were all busy but at a ~50% rate).
My guess is that I did not install/configure PyTorch appropriately, any ideas? Thanks

limbo018, 2019-07-04:
> Hi, would you happen to have any tips on how to properly install PyTorch(+other dependencies) in order to get the best performance out of a CPU-only environment.
> I could not get optimal performance [...]

Hi gessfred, the CPU parallelization is rather experimental yet. Some parts are still implemented in a sequential way, such as DCT. Moreover, it might be better to use a different parallelization scheme for CPU from that for GPU, while currently the same scheme is used with atomic operations. This probably result in severe contention and low efficiency. 

To improve the CPU efficiency, I would suggest to optimize the implementations of the weighted_average_wirelength kernel, electric_density kernel, and DCT kernel. Due to limited bandwidth, it may take some time. Any pull request about such improvement is welcome.

gessfred, 2019-07-06:
Fair enough, thank you :)

closed 2019-07-06 by gessfred