limbo018/DREAMPlace #8: Docker

choltz95, 2019-09-17:
Great paper & thank you for making the project os. Docker would be really nice to make easy installation on windows/osx.

limbo018, 2019-09-21:
> Great paper & thank you for making the project os. Docker would be really nice to make easy installation on windows/osx.

Thanks for the great suggestion. We are exploring the Docker option as well. It should be included within the next two releases.

limbo018, 2019-11-02:
> > Great paper & thank you for making the project os. Docker would be really nice to make easy installation on windows/osx.
> 
> Thanks for the great suggestion. We are exploring the Docker option as well. It should be included within the next two releases.

We have added Docker support in the latest develop branch. You can check out the develop branch and pull the latest version [3e3278a](https://github.com/limbo018/DREAMPlace/commit/3e3278a405fd7489d206f17354efdbe0d694ca57). While we are still waiting for more user feedbacks before a new release, you may want to have a try on the docker environment. Please let us know if the docker works on your side.

closed 2019-11-05 by limbo018