limbo018/DREAMPlace #9: Save intermediate placements

choltz95, 2019-09-24:
Hi, I am interested in recovering intermediate placement positions during global placement. 
I save the initial PlaceDB & I try to get cell placements by accessing parameters in NonLinearPlace model: 
`cur_pos = model.data_collections.pos[0].data.clone().cpu().numpy()
   placement_hist.append(cur_pos)`

When I recover placement_hist after global placement, all positions are 459, but the .pl output yields reasonable placements:
`numpy.nan_to_num(numpy.unique(placement_hist[100])).sum()
   459.0`

Additionally, when I set the plot flag, the first plot looks okay, but I can't open any subsequent plots.

Do you have any recommendation for what I'm trying to do? Thanks.

ZixuanJiang, 2019-09-25:
Many thanks for your feedback. Unfortunately, we cannot reproduce your issue on our end. Could you offer us more details? The results of _bigblue1_ benchmark are shown below for your information.

The expression of `np.nan_to_num(np.unique(cur_pos)).sum()` is non-stationary across iterations. The figures below show the plots of the global placement.

- 0th iteration. The sum is 4656757760.
[image]

- 100th iteration. The sum is 7133514240.
[image]

- 200th iteration. The sum is 7278047744.
[image]

- 300th iteration. The sum is 7060523520.
[image]

- 400th iteration. The sum is 6763920896.
[image]

- 500th iteration. The sum is 6733430272.
[image]

- 600th iteration. The sum is 6738869248.
[image]

- 676th iteration
[image]

choltz95, 2019-09-25:
Hi, thank you very much for your detailed response. - I have found the reasons for my issues.

I don't have a GPU machine at the moment. I am running in a Google colab environment w/ Tesla K80 (compute capability ~3.7).

The reason for the issues is that K80 does not satisfy compute capability requirement described in the README, and build fails if I reduce the cuda cc parameter. 

I am able to recover intermediate placements by running placement on the CPU.

limbo018, 2019-09-28:
> Hi, thank you very much for your detailed response - I have found the reasons for my issues.
> 
> I don't have a GPU machine at the moment. I am running in a Google colab environment w/ Tesla K80 (compute capability ~3.7).
> 
> I guess the reason for the issues is that K80 does not satisfy compute capability requirement described in the README, and build fails if I reduce the cmake cuda cc parameter to 30.
> 
> I am able to recover intermediate placements by running placement on the CPU.
> 
> Also, FYI it looks like PlaceDB object is missing the read_pl function which is called at the end of placement.

Yes, compute compatibility 60 or higher is required for GPU. 

Btw, could you print the error message on the missing read_pl function you mentioned? 

Thanks

choltz95, 2019-09-29:
`Traceback (most recent call last):
  File "dreamplace/Placer.py", line 102, in <module>
    place(params)
  File "dreamplace/Placer.py", line 68, in place
    placedb.read_pl(dp_out_file+".ntup.pl")
AttributeError: 'PlaceDB' object has no attribute 'read_pl'`

limbo018, 2019-09-30:
> ```
> Traceback (most recent call last):
> File "dreamplace/Placer.py", line 102, in <module>
>     place(params)
> File "dreamplace/Placer.py", line 68, in place
>     placedb.read_pl(dp_out_file+".ntup.pl")
> AttributeError: 'PlaceDB' object has no attribute 'read_pl'
> ```

Hi, the latest commit [9d94df8](https://github.com/limbo018/DREAMPlace/commit/9d94df827714356d6d0e98a1c58180881ef66a89) has fixed this issue. You can check it out on master branch. 

Thanks

closed 2019-10-10 by limbo018