limbo018/DREAMPlace #11: Undefined Symbol

ArEsKay3, 2019-11-04:
Hey all,
I'm having problems when trying to build and run DREAMPlace.
Currently I build and then install it in a directory in the project area DREAMPlace/install.
There seems to be some error about an undefined symbol. Where is this defined? If this is from Limbo, I see no Limbo directory in the install directory.
`
DREAMPlace/install/dreamplace/ops/place_io/place_io_cpp.cpython-36m-x86_64-linux-gnu.so: undefined symbol: _ZN9LefParser11LefDataBase23lef_nowireextension_cbkERKNSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEE
`

limbo018, 2019-11-05:
> Hey all,
> I'm having problems when trying to build and run DREAMPlace.
> Currently I build and then install it in a directory in the project area DREAMPlace/install.
> There seems to be some error about [...]

Hi ArEsKay3, 

This is probably related to C++11 and dual ABI settings with gcc. Do you mind switching to the docker container environment which is much more stable for such issues? The latest version 1.1.0 on commit [d09d80a](https://github.com/limbo018/DREAMPlace/commit/d09d8dadd2aa497e2c0f9bc12d4a64acc91e9c92) has added the support. You can follow the instructions in the readme to download/build the environment. Please let me know if it solves the issue.

ArEsKay3, 2019-11-05:
Unfortunately, I'm trying to build this on top of my own container for other purposes, so it's difficult to do.
I'll just try to flip the `cmake -DCMAKE_CXX_ABI=0` flag for the moment. Any other suggestions?

closed 2019-11-05 by ArEsKay3

GuptaPravirSingh, 2020-09-15:
Hi,
I am getting a similar error and changing -DCMAKE_CXX_ABI flag is not helping. Is there any fix for this?