limbo018/DREAMPlace #12: error: DefParser::Region has not been declared

wjx-1999, 2019-12-20:
in latest version,I have a problem that error: DefParser::Region has not been declared,how to solve ?
[image]

limbo018, 2019-12-20:
> in latest version,I have a problem that error: DefParser::Region has not been declared,how to solve ?
> [image]

Hi, have you updated the Limbo submodule with ```git submodule update --recursive``` as well? You can check whether your Limbo version is at commit ec467321410b0904c926d7190d0a327be863e867

Please let me know whether this helps or not. 

Thanks
Yibo

wjx-1999, 2019-12-20:
[image]
I still have problem

limbo018, 2019-12-20:
> [image]
> I still have problem

Could you give more information? I am trying to duplicate your problem. 
1. Is your repo freshly pulled? 
2. Are you using the Docker? 

Thanks

limbo018, 2019-12-21:
I just tried to compile from a freshly pulled repo in both my local environment and the Docker environment. Unfortunately, I was not able to reproduce the error. Please provide more information about your environment.

wjx-1999, 2019-12-21:
i don't use the Docker.Because my computer doesn't have a Nvidia graphics card,it doesn't have CUDA. And then, I annotated a passage from CMakeLists.txt
[image]
[image]

limbo018, 2019-12-21:
Could you also provide your gcc version? It is not shown in the screenshot. I can find a machine without CUDA to test. It is expected to work.

wjx-1999, 2019-12-21:
gcc 5.5 cmake 3.15.4 boost 1.66
[image]
[image]

limbo018, 2019-12-21:
I just tried compiling on a machine without CUDA. It worked fine. I used gcc 5.1, boost 63. I also tried gcc 7.4. I don't think gcc 5.5 will fail. 

Do you mind doing some experiments as follows? 
```
cd thirdparty/Limbo
git status
git log -2
```
I'd like to see the message displayed. 

Thanks
Yibo

wjx-1999, 2019-12-21:
[image]
[image]

My python version is 2.7.12, I don't know that's a problem

limbo018, 2019-12-21:
I just tried compiling with python 2.7 and gcc 5.1. Everything looks fine. Is there any way you can send me your full log? You can send to my email: yibolin at pku.edu.cn

When you run ```make```, please use ```make VERBOSE=1 >& log.txt``` to go to verbose mode and redirect the message to the log file. With more information, I may be able to find some clue. Please remove everything in the build folder and redo the cmake step before you make. 

Thanks
Yibo

wjx-1999, 2019-12-21:
compile successfully after updating the code
but i encounter another problem when running cmd  'python dreamplace/Placer.py test/ispd2015/lefdef/mgc_des_perf_1.json'

[image]

limbo018, 2019-12-21:
That's good. I didn't put the benchmarks in the repo, because they are very large. Currently you can try ispd2005 benchmarks, by downloading using the ispd2005.py script. I will prepare similar script for ispd2015 benchmarks soon.

wjx-1999, 2019-12-21:
I still can't run the ispd05 demo test afetr running cmd 'python dreamplace/Placer.py test/ispd2005/adaptec1.json'
the source file directory as follows benchmarks/ispd2005/adaptec1/adaptec1.aux
[image]
[image]

limbo018, 2019-12-21:
I think you were running from the wrong directory. You need to run from the installation directory, not the repo root directory. 
Suppose you installed DREAMPlace to DREAMPlace/install. Then you need to, 
```
cd install
ls benchmarks # check whether ispd2005 benchmark suites are there
python dreamplace/Placer.py test/ispd2005/adaptec1.aux
```

wjx-1999, 2019-12-21:
[image]
I still can't run the ispd05 demo test as your instruction

limbo018, 2019-12-21:
Sorry, I corrected the typo of *.aux to *.json. Meanwhile, as you don't have gpu, please change the gpu parameter in the json file to 0. 
Are you able to open the input files?

wjx-1999, 2019-12-21:
I can open the input files and set gpu = 0, but can't run

limbo018, 2019-12-21:
what's the error message?

wjx-1999, 2019-12-21:
[ERROR  ] unknown option: for
[INFO   ] reading required
Could not open input file required
[ERROR  ] LEF file parsing failed: required
[ASSERT ] /home/wangsf/workspace/DREAMPlace/dreamplace/ops/place_io/src/place_io.cpp:457: DreamPlace::PlaceDB place_io_forward(const pybind11::list&): Assertion `flag' failed: failed to read input LEF files

limbo018, 2019-12-21:
Hi, I got some luck to reproduce the problem and just fixed it. The problem came from parsing the parameters with default values as string. 

Please pull the latest commit from the master branch. Then, you can go to the build directory and run ```make install``` again. I only changed some python files, so it should be quick. 
After that, you can go to the install directory and test again.

wjx-1999, 2019-12-22:
Hi,now I can run the ispd2005 demo test successfully,but there is a problem when I run the ispd2015 demo test. 
[image]
[image]

limbo018, 2019-12-22:
I found the problem came from reading the DEF file with UNPLACED instances which do not specify the ORIENT. The commit [a0d239a](https://github.com/limbo018/DREAMPlace/commit/a0d239a9aef0f51524b38e20ea528fd7db89f7af) has fixed the issue. 

However, it needs to be noted that currently DREAMPlace can only formally support ISPD2015 benchmarks in detailed placement, as routability and fence regions have not been considered in global placement and legalization. The JSON files provided in the repo also set the ```global_place_flag = 0``` and ```legalize_flag = 0```.

wjx-1999, 2019-12-22:
Thank you !
I will continue to testing and feedback.

ArEsKay3, 2020-01-02:
> compile successfully after updating the code
> but i encounter another problem when running cmd 'python dreamplace/Placer.py test/ispd2015/lefdef/mgc_des_perf_1.json'
> 
> [image]

What exactly did you do to fix this? I'm having the same problem.

closed 2020-01-07 by limbo018