limbo018/DREAMPlace #15: Little issue when I try to run DREAMPlace on Colab

CKLee19800303, 2020-02-12:
Source:

https://github.com/CKLee19800303/DREAMPlace_on_Colab 

When I do "**python dreamplace/Placer.py test/ispd2005/adaptec1.json**"
It shows the error message of 
**NameError: name 'weighted_average_wirelength_cuda_merged' is not defined**

It seems due to ImportError on "weighted_average_wirelength_cuda_sparse"

This error lets the file weighted_average_wirelength.py ignore import dreamplace.ops.weighted_average_wirelength.weighted_average_wirelength_cuda_merged

Hence I switch the order of Line 18 & 19 in https://github.com/limbo018/DREAMPlace/blob/master/dreamplace/ops/weighted_average_wirelength/weighted_average_wirelength.py
to walk around this issue.

But I think my walk-around does not really solve the problem. Just for your information. Thanks.

limbo018, 2020-02-13:
Hi, thanks for raising this issue. I have removed the line to import the *_sparse module, because it has already been deprecated. In the latest version, the error report scheme for module importing is more clear. The script will detect whether the program is built with CUDA and decide whether to import the  CUDA modules. 

You may have a try [1ddae06](https://github.com/limbo018/DREAMPlace/commit/1ddae063d7b78f4b66e371eec9c33d2c633c3c09)

CKLee19800303, 2020-02-16:
Thanks a lot. Thanks for the update.

closed 2020-02-16 by CKLee19800303