limbo018/DREAMPlace #17: docker run bug?

jzh800, 2020-03-18:
docker run -it -v $(pwd):/DREAMPlace limbo018/dreamplace:cuda bash
C:\Program Files\Docker\Docker\resources\bin\docker.exe: invalid reference format: repository name must be lowercase.
See 'C:\Program Files\Docker\Docker\resources\bin\docker.exe run --help'.

 docker run -it -v $(pwd):/dreamlace limbo018/dreamplace:cuda bash
C:\Program Files\Docker\Docker\resources\bin\docker.exe: invalid reference format.
See 'C:\Program Files\Docker\Docker\resources\bin\docker.exe run --help'.

using this command could go into bash shell.
docker run -it -v /dreamlace limbo018/dreamplace:cuda bash

win10 docker hub

limbo018, 2020-03-18:
Do you have a space in your repo name? Can you try change "dreamlace limbo018" to "dreamplace_limbo018"?

jzh800, 2020-03-18:
docker run -it -v dreamplace_limbo018/dreamplace:cuda bash
C:\Program Files\Docker\Docker\resources\bin\docker.exe: Error response from daemon: invalid volume specification: 'dreamplace_limbo018/dreamplace:cuda': invalid mount config for type "volume": invalid mount path: 'cuda' mount path must be absolute.
See 'C:\Program Files\Docker\Docker\resources\bin\docker.exe run --help'.

limbo018, 2020-03-18:
Sorry for the misunderstanding about your command line code. Your original command with a space should be correct. 

If I understand correctly, you suggest the following command and it works under windows. 
docker run -it -v /dreamplace limbo018/dreamplace:cuda bash

That's possible, because I only tested under linux and $(pwd) is not necessary.

jzh800, 2020-03-18:
docker run -it -v /dreamplace limbo018/dreamplace:cuda bash

using this command could go into bash shell.

jzh800, 2020-03-23:
merge in

closed 2020-03-23 by jzh800