limbo018/DREAMPlace #20: Build with Docker: ModuleNotFoundError

cq7522100, 2020-04-03:
Hi, thanks for the open source.
I try to follow the steps in "Build with Docker", 
but in step 6, since there is no directory "DREAMPlace", I do the command:
`git clone --recursive https://github.com/limbo018/DREAMPlace.git`
after getting benchmarks, I try to run it with
`python dreamplace/Placer.py test/ispd2005/adaptec1.json`

but it shows error message:
```
Traceback (most recent call last):
    File "dreamplace/Placer.py", line 21, in <module>
        import PlaceDB
    File "/dreamplace/DREAMPlace/dreamplace/PlaceDB.py", line 16, in <module>
        import dreamplace.ops.place_io.place_io as place_io
    File "/dreamplace/DREAMPlace/dreamplace/ops/place_io/place_io.py", line 9, in <module>
        import dreamplace.ops.place_io.place_io_cpp as place_io_cpp
ModuleNotFoundError: No module named 'dreamplace.ops.place_io.place_io_cpp'
```
I've also tried `make` and `cmake` , but it doesn't work.
Did I miss some steps when trying to build it on Docker?

I appreciate your kind assistance.

limbo018, 2020-04-03:
First, you need to clone the repo. Then you can follow the instructions in the docker section (docker only sets up the building environment). 
According to your error, you are not running from the installation directory. After you make install, go to the install directory so that you can run the program correctly.

cq7522100, 2020-04-03:
Thank you for the prompt rely.
I clone the repo and open docker container from the repo. However, the dreamplace directory is still empty in docker.

limbo018, 2020-04-03:
> Thank you for the prompt rely.
> I clone the repo and open docker container from the repo. However, the dreamplace directory is still empty in docker.

Can you provide more details? 
1. Which OS are you using? 
2. What command did you use to mount the DREAMPlace directory into docker? 
3. And from which directory did you run the command? 
4. Did your docker environment successfully run other tests/tasks? Just to make sure docker is installed correctly.

cq7522100, 2020-04-03:
My OS is Windows10. I clone the repo DREAMPlace at my Desktop and open docker under this path

[image]

I've tried  `docker run hello-world` and `docker run -it ubuntu bash` to make sure docker is working.

limbo018, 2020-04-04:
I don't have a windows machine to test, but according to other users, you may need to use lowercase instead of uppercase. 
https://github.com/limbo018/DREAMPlace/issues/17#issuecomment-600417302

Also, if you take a look at the command on Linux, 
run -it -v $(pwd):/DREAMPlace limbo018/dreamplace:cuda bash

Can you try to replace $(pwd) with the current path on windows? There should be equivalent command on Windows shell. This command means mount this path $(pwd)to /DREAMPlace in the docker environment. So you should be able to change$(pwd) to whatever path you are in.

cq7522100, 2020-04-04:
Thanks for the advice. Adding current path in the command helps to successfully open the DREAMPlace repo in docker.
But then I got the same error message when trying to run it:
[image]

limbo018, 2020-04-05:
Please check this answer https://github.com/limbo018/DREAMPlace/issues/20#issuecomment-608398672. You need to run from the install directory. You are now running from the repo directory...
I assume you have following folder structure 
DREAMPlace 
     - build
     - install 
     - dreamplace
     - test
     - benchmarks
     - unitest
     - CMakeLists.txt
     - README.md

After you run ```make install```, go to the DREAMPlace/install directory before you execute the program.

cq7522100, 2020-04-05:
Okay, so I also have to continue the steps in “Build without Docker” section.
At the ` cmake ` step, here’re some libraries that couldn’t be found: 
[image]
Is it because that Boost are not automatically built?
When I try to continue with `make ` or `make install` command, it’s stuck at the part of 
**[ 92%] Generating place_io.stamp**

limbo018, 2020-04-05:
> Okay, so I also have to continue the steps in “Build without Docker” section.
> At the `cmake` step, here’re some libraries that couldn’t be found:
> [image]
> Is it because that Boost are not [...]

The cmake complaints about these missing libraries are normal, as long as it finishes without error. These warning comes from a thirdparty dependency. DREAMPlace does not need to link boost, so it should be fine. 

It may take long to compile place_io operator, because pybind11 is adopted for python binding, which is fully in C++ templates. If you have any experience template programming, it is long to instantiate all the template classes/functions at compile time.

cq7522100, 2020-04-05:
The problem of getting stuck at 92% fixed after assigning larger memory for docker.
I can now successfully run the code.
Thanks a lot for your help.

closed 2020-04-05 by cq7522100

jeshu54, 2020-05-27:
@limbo018  My terminal gets stuck at make and make install, it's been running for an hour now. There was an internal compiler error as well.
[image]

jeshu54, 2020-05-27:
@cq7522100 Could you please let me know what did you do to overcome the 92%? My code exits with error and no install directory is created. I think I am messing up the directories.

jeshu54, 2020-05-27:
> Please check this answer [#20 (comment)](https://github.com/limbo018/DREAMPlace/issues/20#issuecomment-608398672). You need to run from the install directory. You are now running from the repo directory...
> I assume you have following folder structure
> DREAMPlace/
> - build
> - install
> - dreamplace
> - test
> - benchmarks
> - unitest
> - CMakeLists.txt
> - README.md
> 
> After you run `make install`, go to the DREAMPlace/install directory before you execute the program.

@limbo018 I am not getting the folder structures. Install directory is missing.

limbo018, 2020-05-29:
> @limbo018 My terminal gets stuck at make and make install, it's been running for an hour now. There was an internal compiler error as well.
> [image]

Hi, according to my understanding, cq7522100 solved the making problem in docker by assigning more memory to docker. Binding C++ to python is slow because it relies on pybind11 with a lot of template instantiation. It looks uneasy to build it under windows, but anyway you can check it out.

jeshu54, 2020-05-29:
Thanks Libo Yin for the response.

I am working on Mac. Both Make and Makeinstall process yesterday got finished with the compiler error I shared last time. However, no install folder was created in DREAMPlace directory. Do you still want me to allocate more memory and try again?

Sincerely,
Avro Mukherjee

[Screen Shot 2020-05-27 at 5 40 43 PM]<https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F60354888%2F83075053-38885a80-a041-11ea-9950-3b3e300d8eeb.png&data=02%7C01%7Camukher2%40stevens.edu%7Cd8fb86709a3347c0368008d803cff532%7C8d1a69ec03b54345ae21dad112f5fb4f%7C0%7C0%7C637263538999675823&sdata=9qMnpPsXVG8YE7v1WuggL9ChqGkICruxhKUBl8yxN%2FQ%3D&reserved=0>

limbo018, 2020-05-29:
You will not get any install directory until the code is compiled successfully without any errors, as  making install is executed after successful making. 

You are using the docker under MAC or directly compile with MAC?

jeshu54, 2020-05-29:
I have installed the docker using your instruction on my macbook.

On terminal I went to dream place directory, mkdir build , then cd build.

I used the below code

cmake .. -DCMAKE_INSTALL_PREFIX=my build path directory

Configuration done

I ran make, and got the compile error.

jeshu54, 2020-05-30:
@limbo018 Hi,

I was able to run make and make install successfully without any error. Yes the install directory was not created.

limbo018, 2020-05-31:
> @limbo018 Hi,
> 
> I was able to run make and make install successfully without any error. Yes the install directory was not created.

What is your install directory? Is it absolute path or relative path? What's the message on the screen when you run make install?

jeshu54, 2020-05-31:
@limbo018 
I ran the make on DREAMPlace/build and ran the make install in the same directory.

Please check below screenshots for command line and end result.

Make command 
[image]

Make end result: Final lines

[image]

Make install command:
[image]

Make install output: Final lines

[image]
 
in Mac terminal I ran this code in root directory /DREAMPlace
-->docker run -it -v $(pwd):/DREAMPlace limbo018/dreamplace:cuda bash

then cd /DREAMplace

mkdir build
cd build
cmake .. -DCMAKE_INSTALL_PREFIX=.
make 
make install

both were successfully installed. 

I used the first option from these

[image]

Not sure where I am going wrong.

limbo018, 2020-05-31:
I don't think you can install within the build folder. This is the reserved folder for cmake. Can you try with following path instead?

cmake .. -DCMAKE_INSTALL_PREFIX=../install

jeshu54, 2020-05-31:
This worked. Install directory generated. Thanks so much.