limbo018/DREAMPlace #21: Floating point exception

noeldan, 2020-04-24:
DREAMPlace runs fine when I run with CPU.
However with GPU enabled I am getting this error.

[DEBUG  ] destroying CUDA state takes 2.07456 ms
[DEBUG  ] destroying CPU state takes 0.004025 ms
[INFO   ] Kernel time: 9.22074 ms
[INFO   ] K-reorder time: 97.2032 ms
[INFO   ] Legality check takes 18.2819 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] diamond search grid size 64, sequence length 2048
[INFO   ] initializing GPU memory takes 7.92807 ms
[INFO   ] initial hpwl 0
Floating point exception (core dumped)

Please help me with this.

noeldan, 2020-04-24:
Update
Below are the extended logs.
Kindly help me with this.

[INFO   ] DREAMPlace - Start legalization
[INFO   ] Macro legalization: regard 0 cells as dummy fixed (movable macros)
[INFO   ] Macro legalization takes 0.439973 ms
[DEBUG  ] Standard cell legalization blank_num_bins_y = 890
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 890
[INFO   ] Standard cell legalization legalizeBin takes 2462.944 ms
[DEBUG  ] Standard cell legalization num_unplaced_cells = 0
[DEBUG  ] Standard cell legalization blank_num_bins_y = 890
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 890
[INFO   ] Standard cell legalization legalizeBin takes 1632.484 ms
[DEBUG  ] Standard cell legalization num_unplaced_cells = 0
[INFO   ] Greedy legalization takes 4103.96 ms
[INFO   ] Legality check takes 18.1212 ms
[DEBUG  ] average displace = 2521.69
[INFO   ] Abacus legalization takes 79.1637 ms
[INFO   ] DREAMPlace - legalization takes 4.224 seconds
[INFO   ] DREAMPlace - iteration  101, HPWL 4.422840E+02, time 0.594ms
[INFO   ] DREAMPlace - Start ABCDPlace for refinement
[DEBUG  ] 4-reorder
[DEBUG  ] 24 permutations
[DEBUG  ] initializing CPU DB takes 24.7506 ms
[DEBUG  ] computing independent rows takes 55.3244 ms
[DEBUG  ] computing reorder instances takes 2.04968 ms
[DEBUG  ] initializing CUDA state takes 3.59782 ms
[INFO   ] initial hpwl = 0.000
[INFO   ] Iteration time(ms)     4.49817
[INFO   ] iteration 0: hpwl 0.000 => 0.000 (imp. -nan%)
[INFO   ] Iteration time(ms)     4.50536
[INFO   ] iteration 1: hpwl 0.000 => 0.000 (imp. -nan%)
[DEBUG  ] destroying CUDA state takes 2.07456 ms
[DEBUG  ] destroying CPU state takes 0.004025 ms
[INFO   ] Kernel time: 9.22074 ms
[INFO   ] K-reorder time: 97.2032 ms
[INFO   ] Legality check takes 18.2819 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] diamond search grid size 64, sequence length 2048
[INFO   ] initializing GPU memory takes 7.92807 ms
[INFO   ] initial hpwl 0
Floating point exception (core dumped)

limbo018, 2020-04-24:
Can you provide the CPU log for comparison as well? It is weird that HPWL is 0. I will compute HPWL improvement ratio in detailed placement. That's why you encountered floating point exception.

noeldan, 2020-04-24:
Thank You Dr. Lin for respoding.

Here is the CPU log.

[INFO   ] Greedy legalization takes 503.444 ms
[INFO   ] DREAMPlace - legalization takes 0.561 seconds
[INFO   ] DREAMPlace - iteration  598, HPWL 7.393327E+07, time 8.368ms
[INFO   ] DREAMPlace - Start ABCDPlace for refinement
[INFO   ] Legality check takes 16.7344 ms
[DEBUG  ] average displace = 13.8346
[INFO   ] Abacus legalization takes 36.5088 ms
[DEBUG  ] 4-reorder
[DEBUG  ] compute_independent_rows takes 34.3039 ms
[INFO   ] initial hpwl = 73932864.000
[INFO   ] Iter 0 time (ms) 	 1464.01
[INFO   ] iteration 0: hpwl 73932864.000 => 73647088.000 (imp. 0.386534%)
[INFO   ] Iter 1 time (ms) 	 1452.62
[INFO   ] iteration 1: hpwl 73932864.000 => 73610096.000 (imp. 0.436569%)
[INFO   ] kernel 	 time (ms) 	 runs
[INFO   ] compute_reorder_hpwl 	 8485.43 	 4880976 	 0.00173847
[INFO   ] apply_reorder 	 753.307 	 203374 	 0.00370405
[INFO   ] K-reorder time: 2977.21 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] Legality check takes 19.8545 ms
[INFO   ] diamond search grid size 64, sequence length 2048
[INFO   ] initial hpwl 7.36101e+07
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 526 sets, average set size 105, max set size 128
[INFO   ] iteration 0, target hpwl 7.35169e+07, delta -93200(-0.126613%), solved 526 sets, moved 4.53097% cells, runtime 142.008 ms
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 500 sets, average set size 104, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 501 sets, average set size 103, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 493 sets, average set size 103, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 490 sets, average set size 104, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 492 sets, average set size 103, max set size 128
[INFO   ] iteration 5, target hpwl 7.33272e+07, delta -282880(-0.384295%), solved 492 sets, moved 9.80019% cells, runtime 126.794 ms
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 501 sets, average set size 101, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 500 sets, average set size 101, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 498 sets, average set size 102, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 502 sets, average set size 101, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 494 sets, average set size 103, max set size 128
[INFO   ] iteration 10, target hpwl 7.32483e+07, delta -361840(-0.491563%), solved 494 sets, moved 11.5503% cells, runtime 134.707 ms
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 502 sets, average set size 101, max set size 128
[DEBUG  ] maximal_independent_set_sequential
[DEBUG  ] 510 sets, average set size 100, max set size 128
[DEBUG  ] maximal_independent_set_sequential

limbo018, 2020-04-24:
It looks like the error happened much earlier than detailed placement. 
When you take a look at the HPWL and iterations after legalization, you will find the difference. 
In GPU log 
[INFO ] DREAMPlace - iteration 101, HPWL 4.422840E+02, time 0.594ms
In CPU log 
[INFO ] DREAMPlace - iteration 598, HPWL 7.393327E+07, time 8.368ms

Just to check your CUDA environment. What GPU do you use? 

Can you also send your full logs (both CPU and GPU) to my email at pku.edu.cn? I doubt the problem may happen in global placement.

noeldan, 2020-04-24:
I am using Nvidia Tesla k 80. I will send the logs right away.

limbo018, 2020-04-24:
> I am using Nvidia Tesla k 80. I will send the logs right away.

Aha, this is the problem. I need newer GPUs with compatibility later than 6.0. K80 is too old. In the readme, I put the link to check compatibility of your device.
https://developer.nvidia.com/cuda-gpus

noeldan, 2020-04-24:
Oh sorry. I didn't check that. So technically we cannot get it to run on Tesla K80 as per the requirements.

noeldan, 2020-04-26:
Thank You Dr. Lin for the response. It was very helpful.

closed 2020-04-26 by noeldan