limbo018/DREAMPlace #23: Bookshelf file parsing failed: benchmarks/ispd2005/adaptec1/adaptec1.aux

jeshu54, 2020-06-04:
Hi @limbo018 

When I am running the below code.
[image]

I am getting below error.

[image]

I have re downloaded the benchmark files but still getting this error.

Could you please let me know what the error is?

limbo018, 2020-06-04:
What's the path of your benchmark? Is it under install/benchmarks/ispd2005?

jeshu54, 2020-06-04:
@limbo018 The files got uncompressed to the root directory's benchmark folder not the install's benchmark folder. Should I copy the uncompressed files from root's benchmark to install's benchmark?

jeshu54, 2020-06-04:
@limbo018 I think I will be able to solve it. I was uncompressing data from root directory not install.

Thank you for the assistance. This is my first time building something like this.

limbo018, 2020-06-04:
Yes, the benchmarks should be under install/benchmarks. You may check out the aux_input path in the JSON file.

closed 2020-06-11 by limbo018