limbo018/DREAMPlace #24: How to check the metrics for IO and GP?

jeshu54, 2020-06-14:
Hello @limbo018,

Could you please let me know where I can find the IO(s) metrics as well as the GP metrics in output? and is non-linear placement one of the metrics used in the table 2: Experimental results on ISPD 2005 benchmarks?

Thank you for your time. I am attaching my output log for Adaptec1. If you can just point me which lines I should check for GP and IO that would be really helpful.

Best,
Avro Mukherjee
[log.txt](https://github.com/limbo018/DREAMPlace/files/4777056/log.txt)

limbo018, 2020-06-15:
Initial: iteration 0...
GP: iteration 0 to iteration 603 (everything before "[INFO   ] DREAMPlace - optimizer nesterov takes 175.474 seconds")
LG: "Start Legalization" to "Start ABCDPlace", e.g., iteration 604...
DP: "Start ABCDPlace" to "non-linear placement takes...", e.g., iteration 605

Please take a look at NonlinearPlace.py. You will know where the messages are printed in each step.

jeshu54, 2020-06-15:
Thanks Yibo,

Also, could you tell m about IO metrics that is the table?? Is it the time taken for entire placement? I will check the Nonlinearplacement.py as well.

Best,
Avro Mukherjee

limbo018, 2020-06-15:
Do you mean IO runtime or quality metrics? For quality metrics, that's shown in iteration 0. For IO time, look at this line 
"[INFO   ] DREAMPlace - reading database takes 8.34 seconds".

jeshu54, 2020-06-15:
Yes IO time, thank you. Just for curiosity, where will I find IO metrics? In the table where every metric is mentioned, I am talking about that IO.

Best,
Avro Mukherjee

limbo018, 2020-06-15:
Which table? I got confused. Do you mean any table in a paper?

jeshu54, 2020-06-15:
The IO metric in the attached table.

limbo018, 2020-06-15:
I don't see any attachment. Only the log.txt file you sent in the first message.

jeshu54, 2020-06-15:
I replied through email therefore the file did not got attached. Please find it attached.
[image]

IO(s) column in Dream place section.

limbo018, 2020-06-15:
It consists of two parts
1. DREAMPlace - reading database takes 8.34 seconds
2. [INFO   ] DREAMPlace - write SolutionFileFormat.BOOKSHELF takes 0.642 seconds

The runtime may differ from machine to machine and the running environment (many programmings running on the machine may slow it down). It looks like your machine is slightly slower than the machine I experimented on in the paper.

jeshu54, 2020-06-15:
Understood, the input and output combined. My macbook is 13inch pro, so less power, and also no gpu acceleration. I think this is why my GP timing is very large, sometimes even more than replace metrics.

limbo018, 2020-06-15:
I recommend you to try on a server with recent GPUs, so you will feel the speedup :)

jeshu54, 2020-06-16:
Thanks @limbo018  for the suggestion. Will definitely try it.

Sincerely,
Avro Mukherjee

closed 2020-06-16 by jeshu54