limbo018/DREAMPlace #25: DREAMPlace not handling power pins (VSS, VDD) placed into PINS section

kslobodian, 2020-07-03:
We're currently evaluating viability of DREAMPlace framework by testing it against the sample chip design provided by Mentor Graphics (sample included into Nitro tool starting pack). We found out that DREAMPlace is not able to import power pins (VSS, VDD) placed into PINS section and throws. As an workaround, we removed those pins from the .def file in order to proceed with placement. 

Any plans to support power pins? 

p.s. warnings and final assert message we were getting:

[WARNING] no layer specified by pin VDD, ignored
[WARNING] no layer specified by pin VSS, ignored
[ASSERT ] /DREAMPlace/dreamplace/ops/place_io/src/PyPlaceDB.cpp:312: void DreamPlace::PyPlaceDB::set(const DreamPlace::PlaceDB&): Assertion `num_nodes == node_x.size()' failed: 300557 != 300559

limbo018, 2020-07-04:
Hi, thank you for using DREAMPlace. It looks like an issue related with DEF syntax. Is it possible to have dummy/sample LEF/DEF files and show how the correct ones look like?

svld, 2020-07-06:
Hi, I'm not sure how much I can share at the moment, but I hope the following excerption will add more clarity to the issue.

We have a def file with a preplacement layout where power pins defined in the following way in PINS section:
 - VDD + NET VDD 
      + DIRECTION INOUT
      + USE POWER ; 
   - VSS + NET VSS 
      + DIRECTION INOUT
      + USE GROUND ;
As far is a layer is not specified for such pin, it gets ignored with the warning above and later app asserts.

One more point: I searched through ispd2015 benchmarks. Any on them has power pins declared in PINS section.

limbo018, 2020-07-07:
Thanks for the details. I can reproduce the warnings now, but it didn't fail on my side somehow. 

I found the warning message is triggered at the following line
https://github.com/limbo018/DREAMPlace/blob/24f048f0fe6e000ce1bb2d806058c03ba4ed3d4c/dreamplace/ops/place_io/src/PlaceDB.cpp#L257

I need to know the layer and locations of pins for placement. The following statements are valid. 

- VDD + NET VDD + SPECIAL + DIRECTION INOUT + USE POWER
  + LAYER metal1 ( -25 0 ) ( 25 50 )
  + FIXED ( 0 42400 ) E ;
- VSS + NET VSS + SPECIAL + DIRECTION INOUT + USE GROUND
  + LAYER metal1 ( -25 0 ) ( 25 50 )
  + FIXED ( 0 42200 ) E ;

If you remove the "+ LAYER" statement like following, the warning will be triggered and the pins will be ignored. 

- VDD + NET VDD + SPECIAL + DIRECTION INOUT + USE POWER
  + FIXED ( 0 42400 ) E ;
- VSS + NET VSS + SPECIAL + DIRECTION INOUT + USE GROUND
  + FIXED ( 0 42200 ) E ;

I added a hotfix commit (56a0f52)[https://github.com/limbo018/DREAMPlace/commit/56a0f52600106ea9cfe979635309487e172efe81] that should fix the problem. You may have a try. The warning is still there, but I will no longer ignore the pins. 

Please let me know if this helps. If yes, I will merge it to the master branch. 

Thanks

svld, 2020-07-07:
Hi,
It helped in my configuration.
As for the fail, it asserts here in my case
https://github.com/limbo018/DREAMPlace/blob/24f048f0fe6e000ce1bb2d806058c03ba4ed3d4c/dreamplace/ops/place_io/src/PyPlaceDB.cpp#L294

The reason in such case, I believe, is in incorrect value of num_terminal_NIs at line 228 and, finally, the wrong offset calculated in sub condition "i >= db.nodes().size() - num_terminal_NIs"
To protect yourself you may want to validate DB on whether the number of actually read pins is equal to specified count in PINS section header.

Thank you

limbo018, 2020-07-08:
I uncommented the following line and add print to num_terminal_NIs. 
https://github.com/limbo018/DREAMPlace/blob/24f048f0fe6e000ce1bb2d806058c03ba4ed3d4c/dreamplace/ops/place_io/src/PyPlaceDB.cpp#L292

The numbers are correct on the benchmark I have. I think there might be some other statements in your benchmark triggered the mismatch. It will be hard to debug without the design that can reproduce the problem.