limbo018/DREAMPlace #26: Instructions up to date?

harismeharis, 2020-07-06:
Hi,

Really interesting approach and well written paper. However I am trying to follow closely the instructions to run the benchmarks (python dreamplace/Placer.py test/ispd2005/adaptec1.json) and it is not working. Are the instructions up to date?

harismeharis, 2020-07-08:
To be more specific, I am running : 
 python dreamplace/Placer.py test/ispd2005/adaptec1.json
and I get : 
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 19, in <module>
    import dreamplace.configure as configure
ModuleNotFoundError: No module named 'dreamplace.configure'

limbo018, 2020-07-08:
Please make sure you execute the commands in the *install* directory, not the repo directory.

closed 2020-07-11 by harismeharis