limbo018/DREAMPlace #27: Consider creating demos / tutorials with Google's new open source PDK?

mithro, 2020-07-11:
I don't know if you heard, but @Google has released a manufacturable PDK for a 130nm process under an open source license. You can find it at https://github.com/google/skywater-pdk and see a talk from me on the topic at https://www.youtube.com/watch?v=EczW2IWdnOM -- the docs are slowly getting published at https://skywater-pdk.readthedocs.io/ and there is a public slack channel you can access via https://join.slack.com/t/skywater-pdk/shared_invite/zt-fkl21w8j-qzxBK852XGR8EFMbRakTMw

With a PDK that can now be downloaded with just a git clone (no NDA or other restrictions), do you think it would make sense to create demos and/or tutorials using this PDK so that people can try out DREAMPlace and also reproduce your results?

nihilisticpsycho, 2020-07-13:
@mithro that seems good option I was looking for projects along the newly released PDK too.

limbo018, 2020-07-14:
Thanks for the suggestion. We are searching around for any open-source designs (in DEF format) generated by the skywater PDK so that we can create a demo. OpenRoad is considering to support the new PDK. We may need to wait a little more time for the toolchain ready with the capability of making new benchmarks.

nihilisticpsycho, 2020-07-14:
Please do mention me when you have found out. We can collaborate on making basic demos and brainstorm for novel chip making ideas @limbo018

mithro, 2020-07-14:
@msaligane might already have some.

msaligane, 2020-07-14:
@limbo018 I can help with that. I generated a small testcase using [openroad flow](https://github.com/The-OpenROAD-Project/OpenROAD-flow). I have a DEF already of a gcd design but I am sure I can generate other DEFs for you. Please let me know what you need exactly.

limbo018, 2020-07-15:
> @limbo018 I can help with that. I generated a small testcase using [openroad flow](https://github.com/The-OpenROAD-Project/OpenROAD-flow). I have a DEF already of a gcd design but I am sure I can generate other DEFs for you. Please let me know what you need exactly.

That's great! It would be good if you can generate designs with sizes around 10K - 1M and share the LEF/DEF files for placement.