limbo018/DREAMPlace #28: ERROR in "import dreamplace.ops.place_io.place_io_cpp as place_io_cpp"

chansondog, 2020-08-05:
```
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 21, in <module>
    import PlaceDB
  File "/root/plugins/DREAMPlace01/dreamplace/PlaceDB.py", line 17, in <module>
    import dreamplace.ops.place_io.place_io as place_io 
  File "/root/plugins/DREAMPlace01/dreamplace/ops/place_io/place_io.py", line 9, in <module>
    import dreamplace.ops.place_io.place_io_cpp as place_io_cpp
ModuleNotFoundError: No module named 'dreamplace.ops.place_io.place_io_cpp'
```
It can't find the place_io_cpp. What should i do about it?

chansondog, 2020-08-05:
The command is `python3 dreamplace/Placer.py test/ispd2005/adaptec1.json`
Is it my configuration problem?

limbo018, 2020-08-06:
According to the instructions, you need to run from the **install** directory, not the repo directory.

chansondog, 2020-08-06:
I am sorry for my carelessness.
However, when i run from the install directory, it occurs the same problem of `ModuleNotFoundError: No module named 'dreamplace.ops.place_io'`
So i ls `dreamplace/ops/`, only the `__init__.py  __pycache__`, No floder place_io and others.
I solved this problem by using docker this morning.
May be my system is `DISTRIB_DESCRIPTION="Ubuntu 20.04 LTS"` while the docker is `DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS"`?

closed 2020-08-07 by chansondog

HAOHANZHOU, 2022-01-13:
@sjchanson 
Hi, bro!
I also use docker to build, but I still met the same problem.
Do we need to concern install and repo directory when building with docker?
Thanks.
[image]

limbo018, 2022-01-13:
Yes. You need to run from the install directory, not from the repo directory.