limbo018/DREAMPlace #29: Floating point exception (core dumped)

binderwang, 2020-08-18:
Hi,
First thank you for your code. When I run the example case with "python dreamplace/Placer.py test/ispd2005/adaptec1.json", I got the following error, any suggestions?

========================================================
                       DREAMPlace
            Yibo Lin (http://yibolin.com)
   David Z. Pan (http://users.ece.utexas.edu/~dpan)
========================================================
[INFO   ] DREAMPlace - parameters = {'aux_input': 'benchmarks/ispd2005/adaptec1/adaptec1.aux', 'lef_input': '', 'def_input': '', 'verilog_input': '', 'gpu': 1, 'num_bins_x': 512, 'num_bins_y': 512, 'global_place_stages': [{'num_bins_x': 512, 'num_bins_y': 512, 'iteration': 1000, 'learning_rate': 0.01, 'wirelength': 'weighted_average', 'optimizer': 'nesterov', 'Llambda_density_weight_iteration': 1, 'Lsub_iteration': 1}], 'target_density': 1.0, 'density_weight': 8e-05, 'random_seed': 1000, 'result_dir': 'results', 'scale_factor': 1.0, 'ignore_net_degree': 100, 'gp_noise_ratio': 0.025, 'enable_fillers': 1, 'global_place_flag': 1, 'legalize_flag': 1, 'detailed_place_flag': 1, 'stop_overflow': 0.07, 'dtype': 'float64', 'detailed_place_engine': 'thirdparty/ntuplace3', 'detailed_place_command': '-nolegal -nodetail', 'plot_flag': 0, 'RePlAce_ref_hpwl': 350000, 'RePlAce_LOWER_PCOF': 0.95, 'RePlAce_UPPER_PCOF': 1.05, 'gamma': 4.0, 'RePlAce_skip_energy_flag': 0, 'random_center_init_flag': 1, 'sort_nets_by_degree': 0, 'num_threads': 8, 'dump_global_place_solution_flag': 0, 'dump_legalize_solution_flag': 0, 'routability_opt_flag': 0, 'route_num_bins_x': 512, 'route_num_bins_y': 512, 'node_area_adjust_overflow': 0.15, 'max_num_area_adjust': 3, 'adjust_nctugr_area_flag': 0, 'adjust_rudy_area_flag': 1, 'adjust_pin_area_flag': 1, 'area_adjust_stop_ratio': 0.01, 'route_area_adjust_stop_ratio': 0.01, 'pin_area_adjust_stop_ratio': 0.05, 'unit_horizontal_capacity': 1.5625, 'unit_vertical_capacity': 1.45, 'unit_pin_capacity': 0.058, 'max_route_opt_adjust_rate': 2.0, 'route_opt_adjust_exponent': 2.0, 'pin_stretch_ratio': 1.414213562, 'max_pin_opt_adjust_rate': 1.5, 'deterministic_flag': 0}
[WARNING] DEF input file not specified, cannot output DEF file; set to DEFSIMPLE
[INFO   ] lef_input =
[INFO   ] def_input =
[INFO   ] verilog_input =
[INFO   ] bookshelf_aux_input = benchmarks/ispd2005/adaptec1/adaptec1.aux
[INFO   ] bookshelf_pl_input =
[INFO   ] def_size_input =
[INFO   ] def_output = -out.def
[INFO   ] rpt_output =
[INFO   ] target_util = 0
[INFO   ] max_displace = 0
[INFO   ] bin size = (10, 10) #rows
[INFO   ] sbin size = (5, 5) #rows
[INFO   ] bin_space_threshold = 0.2
[INFO   ] abu = 2,10 5,5 10,2 20,1
[INFO   ] def_ignore_cells =
[INFO   ] macro_obs_aware_layers =
[INFO   ] enable_place = true
[INFO   ] enable_legalize = true
[INFO   ] evaluate_overlap = false
[INFO   ] move_multi_row_cell = true
[INFO   ] align_power_line = true
[INFO   ] cluster_cell = false
[INFO   ] sort_nets_by_degree = false
[INFO   ] file_format = BOOKSHELF
[INFO   ] max_iters = 6
[WARNING] no DEF file specified
[WARNING] no Verilog file specified
[INFO   ] reading benchmarks/ispd2005/adaptec1/adaptec1.aux
[WARNING] no additional Bookshelf .pl file specified
[WARNING] 20285 nets with 24352 pins from same nodes
[WARNING] 4210 nets should be ignored due to not enough pins
[INFO   ] sort nodes in the order of movable and fixed
[INFO   ] group cells for fence regions
[INFO   ] fence region done
[DEBUG  ] num_terminals 543, numFixed 543, numPlaceBlockages 0, num_terminal_NIs 0
[DEBUG  ] fixed area overlap: 4.91641e+07 total: 6.4094e+07, space area = 6.50265e+07
[INFO   ] DREAMPlace - scale coordinate system by 1
[INFO   ] DREAMPlace -
================================= Benchmark Statistics =================================
#nodes = 211447, #terminals = 543, # terminal_NIs = 0, #movable = 210904, #nets = 221142
die area = (459, 459, 11151, 11139) 1.14191e+08
row height = 12, site width = 1
num_bins = 512x512, bin sizes = 1.74023x1.73828
#pins = 919701, #movable_pins = 905028
total_movable_node_area = 3.72863e+07, total_fixed_node_area = 4.91641e+07, total_space_area = 6.50265e+07
target_density = 1
use placeable_area = 6.50265e+07 to compute fillers
total_filler_node_area = 2.77402e+07, #fillers = 160067, filler sizes = 14.442x12
========================================================================================
[INFO   ] DREAMPlace - reading benchmark takes 4.74538 seconds
[INFO   ] DREAMPlace - reading database takes 4.75 seconds
[INFO   ] DREAMPlace - move cells to the center of layout with random noise
[WARNING] DREAMPlace - net weights are all the same, ignored
[INFO   ] DREAMPlace - non-linear placement initialization takes 2.10 seconds
[INFO   ] DREAMPlace - 512x512 bins #bins 512x512, bin sizes 1.74023x1.73828, max_num_bins = 6, padding = 0
[INFO   ] dreamplace.ops.electric_potential.electric_potential - regard 0 cells as movable macros in global placement
[INFO   ] DREAMPlace - use nesterov optimizer
[INFO   ] DREAMPlace - add 2.5% noise
[INFO   ] dreamplace.ops.electric_potential.electric_potential - fixed density map: average 0, max 0, bin area 435.602
[INFO   ] DREAMPlace - nesterov initialization takes 0.0201526 seconds
[... 337 lines cut ...]
[INFO   ] Abacus legalization takes 68.3133 ms
[INFO   ] DREAMPlace - legalization takes 3.381 seconds
[INFO   ] DREAMPlace - iteration  102, HPWL 0.000000E+00, time 0.268ms
[INFO   ] DREAMPlace - Start ABCDPlace for refinement
[DEBUG  ] 4-reorder
[DEBUG  ] 24 permutations
[DEBUG  ] initializing CPU DB takes 20.5197 ms
[DEBUG  ] computing independent rows takes 41.7529 ms
[DEBUG  ] computing reorder instances takes 1.57007 ms
[DEBUG  ] initializing CUDA state takes 2.97754 ms
[INFO   ] initial hpwl = 0.000
[INFO   ] Iteration time(ms)     3.40255
[INFO   ] iteration 0: hpwl 0.000 => 0.000 (imp. -nan%)
[INFO   ] Iteration time(ms)     3.41659
[INFO   ] iteration 1: hpwl 0.000 => 0.000 (imp. -nan%)
[DEBUG  ] destroying CUDA state takes 1.87586 ms
[DEBUG  ] destroying CPU state takes 0.000522 ms
[INFO   ] Kernel time: 6.9565 ms
[INFO   ] K-reorder time: 75.7488 ms
[INFO   ] Legality check takes 14.5999 ms
[INFO   ] DREAMPlace - K-Reorder legal flag = 1
[INFO   ] diamond search grid size 64, sequence length 2048
[INFO   ] initializing GPU memory takes 5.68425 ms
[INFO   ] initial hpwl 0
Floating point exception (core dumped)

limbo018, 2020-08-18:
Hi, I saw a lot of NAN on the messages. You are using the GPU version. How about the CPU version? You can modify the JSON and turn the gpu flag to 0. 

In general, if the CPU version works fine, then the problem comes from your GPU device. It might be too old to be supported. In the README, I have a link pointing to the GPU architecture supported (https://developer.nvidia.com/cuda-gpus). We need at least compatibility 6.0. You may check previously closed issues for more details.

binderwang, 2020-08-18:
> Hi, I saw a lot of NAN on the messages. You are using the GPU version. How about the CPU version? You can modify the JSON and turn the gpu flag to 0.
>
> In general, if the CPU version works fine, then [...]

Well, I see. Thank you very much, I've tested with CPU and it works well. Thanks again for your work.

closed 2020-08-18 by binderwang