limbo018/DREAMPlace #30: Hypergraph Details

jeshu54, 2020-08-25:
@limbo018 Hello,

I want to understand how I can check the hypergraph edges and nodes of any design. I want to check the mapping and how we can implement our hyper graph using dream place. Can you please tell me where should I start looking?

Thanks,
Jeshu

jeshu54, 2020-08-26:
@limbo018 
So far I was able to find the files that contains the nodes and nets of the design, however, I am not able to understand how I can translate it to a sparse matrix. Also if I have my own design how I can map the nodes and nets. Could you please help me here?

limbo018, 2020-08-27:
Here is the description about the bookshelf format. 
http://vlsicad.eecs.umich.edu/BK/ISPD06bench/BookshelfFormat.txt

If you want to directly access the hypergraph, you can check following members in PlaceDB.py

## storing information of pins in each node ##
flat_node2pin_map
flat_node2pin_start_map # these two arrays are a flat version of vector<vector<int>>
pin2node_map # map pin id to its node id 

## storing information of pins in each net ##
flat_net2pin_map
flat_net2pin_start_map # these two arrays are a flat version of vector<vector<int>>
pin2net_map # map pin id to its net id 

But it is safer to convert your design to bookshelf format, because direct modifying the data structures may cause unexpected behavior especially when you are not familiar with the code.

jeshu54, 2020-08-27:
Thanks @limbo018 
How can I access the variable? I am running the code via terminal. So I am not sure how can I access these variables?

jeshu54, 2020-08-27:
@limbo018 I am trying to call the function from placeDB.py, flatten_nested_map which is returning lat_net2pin_map, flat_net2pin_start_map.

I am trying to call the function in placer.py using the below command.

[image]

Can you tell me what parameter I need to pass here in the function flatten_nested_map ?
Thanks,
Avro

jeshu54, 2020-08-28:
@limbo018 could you please share any update? 

Thanks,
Avro

limbo018, 2020-08-28:
I'm not able to load the images on my side. Actually, in the latest version, I have commented out  the function flat_nested_map, which is not used anywhere. If you really want to use it, you need to make sure you have the capability of making the code work. As I mentioned, it will be easier for you to generate bookshelf rather than directly modifying the variables. 

If you insist on doing that (I do not guarantee it will work without huge modification), the argument of net2pin_map is a nested array data structure, i.e., something like vector<vector<int>> in C++. You probably need to modify the function a little bit. I remember there is a small bug (I cannot recall for now), but I didn't fix as the function is not used.

jeshu54, 2020-08-28:
@limbo018  Thank you. Yes I have been constantly facing errors while trying to run the function. So is there any other way I can access the variable data to see the arrays? I am going to use the bookshelves like you mentioned. That will be more efficient. I just wanted to understand how the code is storing the variables and wanted to check what is stored in node2pin and net2pin variables?

limbo018, 2020-08-28:
You can print the variables. It’s in python.

jeshu54, 2020-08-28:
@limbo018  So you mean just use print(placedb.flat_net2pin_start_map) in placer.py and I should be able to get the results.

limbo018, 2020-08-29:
Yes.

closed 2020-09-19 by limbo018