limbo018/DREAMPlace #31: Which GPU card do you recommend?

fsword73, 2020-08-31:
Especially the local video memory size . Does it need 32 GB or 11 GB only? How about System DDR size too?

limbo018, 2020-08-31:
It depends on your design sizes. In general, 11GB is enough for million-gate designs. If you have designs with more than 10 millions of gates, then it requires more than 11GB memory. 

You have to choose GPU cards with later architectures, such as Pascal, Volta, Turing, etc., as we have requirement to the compute compatibility. We have mentioned that in the README as follows, 

GPU architecture compatibility 6.0 or later 

Code has been tested on GPUs with compute compatibility 6.0, 7.0, and 7.5.
Please check the compatibility of the GPU devices (https://developer.nvidia.com/cuda-gpus). 
The default compilation target is compatibility 6.0. This is the minimum requirement and lower compatibility is not supported for the GPU feature.

fsword73, 2020-08-31:
thanks very much!

closed 2020-08-31 by fsword73