limbo018/DREAMPlace #33: how

hhngw, 2020-10-29:


limbo018, 2020-10-29:
> 1. thirdparty/ntuplace3

Did you modify the JSON file and match the path of ntuplace3 with your path, e.g., init/thirdparty/ntuplace3, or whatever else?

For ntuplace4_dr, you need to contact the authors of the NTUplace4dr paper.

hhngw, 2020-10-29:
I have run from the install directory ,the path is  correct  , your script can find the ntuplace engine ,but occor error 
the log show : thirdparty/ntuplace3: 1:thirdparty/ntuplace3 : Syntax error : word unexpected ( expecting ")") ,

limbo018, 2020-10-29:
You can try to run ntuplace3 standalone by directly feeding the input files required. Probably the benchmarks contain syntaxes that cannot be recognized by ntuplace3. Can you open the input files and check the lines report problems?

hhngw, 2020-10-31:
I have run ntuplace3 standalone by directly in ubuntu18.04 terminal ：
sudo thirdparty/ntuplace3 -aux benchmarks/ispd2005/xx/xx.aux  -loadpl /results/xx/xx.pl  -out xx -noglobal -noleagal -nodetail
the log show : thirdparty/ntuplace3: 1:thirdparty/ntuplace3 : Syntax error : word unexpected ( expecting ")")

limbo018, 2020-11-04:
Could you try running the binary by replacing the xx.pl with the original xx.pl? Like following, 
thirdparty/ntuplace3 -aux benchmarks/ispd2005/xx/xx.aux -loadpl benchmarks/ispd2005/xx/xx.pl -out xx 

If it works, then it means there is syntax mismatch in the dumped xx.pl. Otherwise, something might be wrong with the ntuplace3 binary.

hhngw, 2020-11-05:
It  cannot work when I run by the ispd2005  file ，the binary has been compiled in 2012 . maybe incompatible with ubuntu18.04 or gcc7.5
Thank you~

closed 2020-11-05 by hhngw

ybj-catherine, 2024-09-10:
> hi, I have download ntuplace3 bin file and copy to mydir init/thirdparty/ but :
> 
> 1. when i run  ispd2005/adaptec1.json , the log show  : thirdparty/ntuplace3:  1:thirdparty/ntuplace3 : Syntax error : word unexpected ( expecting ")")  ,
> 2. when i run ispd2015/lefdef/mgc_superblue19.json  , the log show  : thirdparty/ntuplace_4dr or aux file NOT found.
>    where down load ntuplace_4dr?
> 
> other steps of dreamplace can be run correct
> 
> tks

hi, where did you download the ntuplace3 bin file? I also need detailed placement but did not find a proper file assinged to params.detailed_place_engine shown in the code

limbo018, 2024-09-10:
Hi, the binaries for ntuplace3 and ntuplace_4dr are not in the repo, because these binaries are not developed by us and we do not have the permission to release them. You may need to ask the original authors of these work to obtain the binaries, i.e., Prof. Yao-Wen Chang's group. 

                       
Original

ybj-catherine, 2024-09-10:
thank you for reply!

Dy233-hub, 2024-12-02:
hi, I also have the problem about finding ntuplace3 bin file. Could you tell me where did you download the ntuplace3 bin file? I also need detailed placement but I can't find the way to get ntuplace3.

limbo018, 2024-12-02:
ntuplace3 is an optional external detailed placer. We did not get the permission from the original authors to release it. You can turn on ```detailed_place_flag``` (set to 1), and empty the ```detailed_place_engine``` and ```detailed_place_command``` fields. This configuration will invoke the default ABCDPlace engine for detailed placement.

Dy233-hub, 2024-12-02:
Thank you,I'll try it later.