limbo018/DREAMPlace #34: Questions on updating to Latest CUDA/cub/thrust

ArEsKay3, 2020-12-09:
Hi all,
    I'm attempting to integrate DREAMPlace with some other code I have which requires CUDA11 which comes with a more updated version of thrust/cub.
    In the process I'm currently debugging two issues related to the detail place step and I was wondering if someone might have any insight.

-- When updating to CUDA11 thrust throws the below error at this line. I'm talking to thrust developers who think they may have fixed this issue, so I'm trying the latest thrust/cub.
https://github.com/limbo018/DREAMPlace/blob/0035d8a8a40729d414c84d52464b459d46680db9/dreamplace/ops/global_swap/src/global_swap_cuda_kernel.cu#L1275 
 `RuntimeError: after reduction step 1: cudaErrorInvalidDevice: invalid device ordinal`

-- With the latest versions of thrust/cub from github, I'm getting an error when doing the D2H copy at the end of partition_kmeans. Specifically it seems I get a memory error when copying `state.selected_maximal_independent_set`.
`dreamplace/ops/independent_set_matching/src/kmeans.cuh: void DreamPlace::kmeans_collect_sets_cuda2cpu(const DetailedPlaceDB
Type&, IndependentSetMatchingStateType&, DreamPlace::KMeansState<typename DetailedPlaceDBType::type>&) [with DetailedPlaceDBType = DreamPlace::DetailedPlaceDB<float>; IndependentSetMatchingStateType = DreamPlace::IndependentSetMatchingState<float>; typename DetailedPlaceDBType::type = float]: Assertion cudaMemcpy(selected_nodes.data(), state.selected_maximal_independent_set, sizeof(int)*state.num_selected, cudaMemcpyDeviceToHost) == cudaSuccess' failed: CUDA Runtime Error: an illegal memory access was encountered`
https://github.com/limbo018/DREAMPlace/blob/0035d8a8a40729d414c84d52464b459d46680db9/dreamplace/ops/independent_set_matching/src/kmeans.cuh#L310
This code (which provides the state.selected_maximal_independent_set) looks like it's using cub::DeviceScan::ExclusiveSum in an unconventional way (using temp storage as results). I was wondering if someone could explain what's going on here so I can see if changes to cub are causing this.
https://github.com/limbo018/DREAMPlace/blob/0035d8a8a40729d414c84d52464b459d46680db9/dreamplace/ops/independent_set_matching/src/select.cuh#L24

limbo018, 2021-06-24:
Hi, sorry for the long delay. I have looked into the problem recently. Although I tried to make global placement compatible to pytorch 1.8, detailed placement still crashed. 

It looks like a common problem when using Thrust and CUB under CUDA 11. The same problem has been reported for PyTorch. 

https://github.com/pytorch/pytorch/issues/52663

I found that the program will definitely crash if we call whatever reduce functions in thrust, even with newly allocated arrays.

limbo018, 2021-06-25:
Hi, I found a workaround to fix the problem by 

1. Put cub into custom namespace. 
2. Replace all thrust call with cub. 

Here is the original discussion about this workaround. 
https://github.com/dmlc/dgl/pull/2758/commits/fd183c156b063c2c783e0a9d3501bce4a44f9aeb

I have also fixed some other bugs in the DP algorithm with the latest cub under CUDA 11. It should work with PyTorch 1.8 + CUDA 11 now.

limbo018, 2021-06-25:
> Hi, sorry for the long delay. I have looked into the problem recently. Although I tried to make global placement compatible to pytorch 1.8, detailed placement still crashed.
> 
> It looks like a common problem when using Thrust and CUB under CUDA 11. The same problem has been reported for PyTorch.
> 
> [pytorch/pytorch#52663](https://github.com/pytorch/pytorch/issues/52663)
> 
> I found that the program will definitely crash if we call whatever reduce functions in thrust, even with newly allocated arrays.

See commit: https://github.com/limbo018/DREAMPlace/commit/32dc2389bc49e67965fc2b0684d3f4e0c1890cb5

limbo018, 2021-06-25:
> Hi, I found a workaround to fix the problem by
> 
> 1. Put cub into custom namespace.
> 2. Replace all thrust call with cub.
> 
> Here is the original discussion about this workaround.
> [dmlc/dgl@fd183c1](https://github.com/dmlc/dgl/commit/fd183c156b063c2c783e0a9d3501bce4a44f9aeb)
> 
> I have also fixed some other bugs in the DP algorithm with the latest cub under CUDA 11. It should work with PyTorch 1.8 + CUDA 11 now.

https://github.com/limbo018/DREAMPlace/commit/32dc2389bc49e67965fc2b0684d3f4e0c1890cb5

PR #41 (2021-07-02): Compatibility patch for CUDA Toolkit 11.0 and PyTorch 1.8
The changes in CMakeLists.txt is compatibility patch for CUDA Toolkit 11.0. Since 11.0, `cub` is already included in CUDA Toolkit, using custom `cub` will trigger an error off. This may fix #34.
For usage of `${CUDA_INCLUDE_DIRS}`, `${CUDA_VERSION_MAJOR}`, and `VERSION_LESS` refer to https://cmake.org/cmake/help/latest/module/FindCUDA.html

The changes in dreamplace/ops is compatibility patch for PyTorch 1.8.
For changes in torch.h the definition of function `rfft()` and `irfft()`, PyTorch 1.7 and 1.8 implements an `at::fft` namespace i [... cut]
closed 2021-07-02 by limbo018