limbo018/DREAMPlace #35: How do I enable macro placement and legalization in DREAMPlace 2.0?

ayeung-fw, 2021-01-26:
I would like to try “macro placement” in DREAMPlace 2.0,
Since non-CORE cells need to be either FIXED or PLACED, I placed the memories at (0 0) as a starting point:
[image]

But that did not work.
[image]

Is there some config that I need to use to enable macro placement?

limbo018, 2021-02-27:
Hi, sorry for late response. I didn't try macro placement on LEF/DEF format before (only tried on bookshelf format), so it took me sometime to figure out the reason. 

I checked the implementation and found that the placer currently only work on CORE cells and always expect non-CORE cells to be fixed. Such a settings is mostly due to the complexity of combining CORE-cells and non-CORE cells. 

To enable macro placement, you need to change the LEF file a little bit like following (an example from ISPD 2015 benchmark): 
```
# cells.lef
...
MACRO h0
   CLASS BLOCK ; # change to CLASS CORE
   FOREIGN h0 ;
   ORIGIN 0 0 ;
   SIZE 329.6 BY 158 ;
   SYMMETRY X Y R90 ;
...
```
```
# floorplan.def
...
   - h0 h0
      + FIXED ( 456600 641205 ) N ; # change to + PLACED ( 456600 641205 ) N
...
```

In this way, it will regard fixed macros as movable macros. You will see a message like following on the screen after printing the benchmark statistics: 
```
[INFO   ] dreamplace.ops.electric_potential.electric_potential - regard 1 cells as movable macros in global placement
```

The macro placement engine is not very stable for now. I observe sometimes it may diverge and the macro legalization may fail when there are too many movable macros. We are working on improving them.

ayeung-fw, 2021-02-27:
Hi Yibo,

Thanks for confirmation my observation.
I was able to figure this out and have been the same workaround as well.

Regards,
Adrian

kangnam4123, 2023-12-04:
Hi, I also want to proceed with macro & Standard cell placement using iccad2015.ot bench. As advised earlier, I changed CLASS to CORE in the lef file, and in the def file, I changed FIXED to PLACED. However, the following log was outputted.
[image]

It seems like the placement is proceeding, but is it correct that the placement is happening properly, or is there something else that needs to be modified?

limbo018, 2023-12-05:
Hi, the log looks ok to me.

kangnam4123, 2023-12-05:
Thank you for fast reply.
On the log, it said there are 0 cells as movable macro in global placement. 
Is it still okay?

limbo018, 2023-12-05:
That's weird. Could you share your case?

kangnam4123, 2023-12-05:
I used iccas2015.ot bench and json file to run dreamplace4.0 version. Additionally, I try to do mixed size placement. So I modify lef,def file as you mentioned above(Fix all Block, Fixed into Core,Placed). But the log said there are no movable macro block while doing placement.
Is there anything to be fixed?
V/R