limbo018/DREAMPlace #37: How do I read a netlist with gates which have multiple output pins?

PeiyanFlying, 2021-04-28:
Hi, I meet with some trouble when I read my own netlist.
This is my own device, a two output spliter. And here is the cell.lef of the spliter2. How ever I meet with this trouble when I run the Global Placement part. Beg your help! Many thanks!

[image]
[image]
[image]

limbo018, 2021-04-28:
According to the error message, I think you need to check the macro named "add_bb" in your LEF file. It complains failed to find a pin named "o".

PeiyanFlying, 2021-04-28:
Thank you for your timely reply. And as you see, there is no pin "o" in the macro "add_bb" defined in my LEF file. 
[image]

PeiyanFlying, 2021-04-28:
And is it a problem in "after_legalized.ntup.fix.def" ? I just use a file from other designs in ispd2015. Because we only need the Global Placement process.
Thanks!

PeiyanFlying, 2021-04-28:
And this is how my netlist look like. Maybe I use the different datatype? (wire, not input?)
[image]
[image]

limbo018, 2021-04-29:
Looks like you have solved your previous problem. 

For the wire type (wire or input/output), the declaration of input/output should match ```adder (a, b, c, cout, s)```. Other signals should be ```wire```.

PeiyanFlying, 2021-05-24:
Yes! Many thanks for your detail reply.

closed 2021-05-24 by PeiyanFlying