limbo018/DREAMPlace #39: meet error: No module named 'dreamplace.ops.place_io.place_io_cpp'

worry1114, 2021-05-21:
i have done cmake but when i run the commond , i meet the error that:

root@79bb4f64a719:/d/dreamplace/DREAMPlace# python dreamplace/Placer.py test/ispd2005/adaptec1.json
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 21, in <module>
    import PlaceDB
  File "/d/dreamplace/DREAMPlace/dreamplace/PlaceDB.py", line 17, in <module>
    import dreamplace.ops.place_io.place_io as place_io 
  File "/d/dreamplace/DREAMPlace/dreamplace/ops/place_io/place_io.py", line 9, in <module>
    import dreamplace.ops.place_io.place_io_cpp as place_io_cpp
ModuleNotFoundError: No module named 'dreamplace.ops.place_io.place_io_cpp'

thanks u very much~

limbo018, 2021-05-21:
You need to run from the install directory, not the root directory of the repo.

TheBigFour, 2022-03-16:
Sorry but I met the same error. I checked what you said and found I did it right. It  seems that the file folder ''dreamplace/ops'' is empty and contains no file named 'dreamplace.ops.place_io.place_io_cpp'. Thanks for your attention!

limbo018, 2022-03-16:
Does your building process succeed? ```make install``` should install stuff into the installing directory, so dreamplace/ops should not be empty. You may check whether there are errors in the building process.

TheBigFour, 2022-03-16:
Now I am trying "building with docker", but the "7.Go to next section to complete building." is hard to understand. I have run the instruction "docker run -it -v $(pwd):/DREAMPlace limbo018/dreamplace:cuda bash", but how to run the code?

zhengsizuo, 2023-06-01:
> Now I am trying "building with docker", but the "7.Go to next section to complete building." is hard to understand. I have run the instruction "docker run -it -v $(pwd):/DREAMPlace limbo018/dreamplace:cuda bash", but how to run the code?

I meet the same problem too, and I am also confused about  "7.Go to next section to complete building."

limbo018, 2023-06-01:
Hi, docker only provides an environment to build the code. It means that after you enter docker's environment, you still need to go through the building process before running the code. The "next section" means to following the instructions of Section "Build without Docker" from beginning to the end.

zhengsizuo, 2023-06-02:
> Owner

Thanks for your replay. I've completed the building process, but I meet the same problem again.
[image]
Maybe because I haven't built Boost manually? "Third party submodules are automatically built except for Boost"

limbo018, 2023-06-02:
Can you try ```ls -l``` and see what's inside the folder? You may still need to go to the ```install``` directory to run the program. The source code directory is different from the install directory.

magic3007, 2023-06-03:
From the screenshot, it appears that you are running under the directory `/DREAMPlace`. You should run the program under `your_install_path`, which is the location specified when running cmake.
[image]

kienprojects, 2024-09-15:
Sorry but I met the same error. I checked what you said and found I did it right. It seems that the file folder ''dreamplace/ops'' is empty and contains no file named 'dreamplace.ops.place_io.place_io_cpp'. Thanks for your attention! This is the output of commands: 
(pl) (base) tludemo@admin1:~/hinhnv/kienle/DeepPlace$ mkdir build 
cd build 
(pl) (base) tludemo@admin1:~/hinhnv/kienle/DeepPlace/build$ cmake .. -DCMAKE_INSTALL_PREFIX=/home/tludemo/hinhnv/kienle/DeepPlace -DPYTHON_EXECUTABLE=$(which python)
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 11.4.0
-- The CXX compiler identification is GNU 11.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) in CMakeLists.txt:
  No cmake_minimum_required command is present.  A line of code such as

    cmake_minimum_required(VERSION 3.22)

  should be added at the top of the file.  The version specified may be lower
  if you wish to support older CMake versions for this project.  For more
  information run "cmake --help-policy CMP0000".
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    PYTHON_EXECUTABLE

-- Build files have been written to: /home/tludemo/hinhnv/kienle/DeepPlace/build
(pl) (base) tludemo@admin1:~/hinhnv/kienle/DeepPlace/build$ make 
[ 50%] Building CXX object ops/utility/CMakeFiles/utility.dir/src/msg.o
[100%] Linking CXX static library libutility.a
[100%] Built target utility
(pl) (base) tludemo@admin1:~/hinhnv/kienle/DeepPlace/build$ make install
Consolidate compiler generated dependencies of target utility
[100%] Built target utility
Install the project...
-- Install configuration: ""
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/BasicPlace.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/EvalMetrics.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/NesterovAcceleratedGradientOptimizer.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/NonLinearPlace.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/Params.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/PlaceDB.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/PlaceObj.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/Placer.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/configure.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/evaluation.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/fullplace_env.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/generate_tmux_yaml.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/main.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/params.json
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/place_env.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/rnd.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/validation.py
-- Installing: /home/tludemo/hinhnv/kienle/DeepPlace/dreamplace/ops/__init__.py

andrecavalcante, 2025-04-18:
I have have the same problem.

limbo018, 2025-04-18:
What's your cmake version? It seems like your cmake does not support the syntax in the CMakeLists.txt.