limbo018/DREAMPlace #43: fatal error: cub/cub.cuh: No such file or directory

RecursionSheep, 2021-07-08:
Hello. I tried to build DREAMPlace without using docker, but I met an error while running `make`:
```
/usr/local/cuda/bin/nvcc -M -D__CUDACC__ /home/recursion/DREAMPlace/dreamplace/ops/pin_pos/src/pin_pos_cuda_segment_kernel.cu -o /home/recursion/DREAMPlace/build/dreamplace/ops/pin_pos/CMakeFiles/pin_pos_cuda_segment_cuda_tmp.dir/src/pin_pos_cuda_segment_cuda_tmp_generated_pin_pos_cuda_segment_kernel.cu.o.NVCC-depend -ccbin /usr/bin/cc -m64 -DTORCH_EXTENSION_NAME=pin_pos_cuda_segment -DTORCH_MAJOR_VERSION=1 -DTORCH_MINOR_VERSION=8 -DENABLE_CUDA=1 -D_GLIBCXX_USE_CXX11_ABI=0 -Xcompiler ,\"-O3\",\"-DNDEBUG\" -gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 --compiler-options -fPIC -DNVCC -I/usr/local/cuda/include -I/home/recursion/DREAMPlace/dreamplace/ops/pin_pos/.. -I/usr/local/lib/python3.8/dist-packages/torch/include -I/usr/local/lib/python3.8/dist-packages/torch/include/torch/csrc/api/include -I/home/recursion/DREAMPlace/thirdparty/pybind11/include -I/usr/include/python3.8
In file included from /home/recursion/DREAMPlace/dreamplace/ops/pin_pos/src/pin_pos_cuda_segment_kernel.cu:6:0:
/home/recursion/DREAMPlace/dreamplace/ops/pin_pos/../utility/src/utils_cub.cuh:17:10: fatal error: cub/cub.cuh: No such file or directory
 #include "cub/cub.cuh"
          ^~~~~~~~~~~~~
compilation terminated.
```
I'm sure I've already cloned the CUB dependency. It seems there's sth wrong with the CMake file?
Thanks for your help!

limbo018, 2021-07-08:
Fixed in commit https://github.com/limbo018/DREAMPlace/commit/8de021d092bacf5264d84081afe6c8a99cf40c4d

RecursionSheep, 2021-07-08:
> Fixed in commit [8de021d](https://github.com/limbo018/DREAMPlace/commit/8de021d092bacf5264d84081afe6c8a99cf40c4d)

Thank you!

closed 2021-07-08 by RecursionSheep