limbo018/DREAMPlace #45: meet error when run the commond: $ cmake .. -DCMAKE_INSTALL_PREFIX=your_install_path for newest version

zchen71, 2021-07-17:
Hi,

I ran into errors when I tried to build files for the newest version in the docker.

"CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.14 or higher is required.  You are running version 3.8.2

-- Configuring incomplete, errors occurred!"

But the cmake in docker works well when I ran previous version. (Only not work for the newest pull) Is it necessary to have CMake 3.14 or higher in the docker to run the newest version?

Thanks

limbo018, 2021-07-18:
Hi,

In the latest version, we have upgraded the CMake version to 3.14 and PyTorch to 1.6. The docker is not updated yet. We still need some time to adjust the docker. You may update the docker on your side while waiting for our updates.

Thanks
Yibo

closed 2021-07-19 by zchen71

limbo018, 2021-07-22:
Hi, commit https://github.com/limbo018/DREAMPlace/commit/684117fa3cca49d50a98f4a8cd6cd73321a173e6 has fixed the issue by updating the docker configuration.