limbo018/DREAMPlace #47: meet error when run cmake .. -DCMAKE_INSTALL_PREFLIX = ...

additionLilian, 2021-09-01:
Hi,

When I run cmake .. -DCMAKE_INSTALL_PREFIX=your_install_path -DPYTHON_EXECUTABLE=$(which python), I meet the following error: 

CMake Error at thirdparty/pybind11/tools/FindPythonLibsNew.cmake:133 (message):
  Python config failure:

Call Stack (most recent call first):
  thirdparty/pybind11/tools/pybind11Tools.cmake:45 (find_package)
  thirdparty/pybind11/tools/pybind11Common.cmake:205 (include)
  thirdparty/pybind11/CMakeLists.txt:190 (include)

-- Configuring incomplete, errors occurred!

limbo018, 2021-09-01:
Hi, I haven't seen this error before. Does it work if you remove ```-DPYTHON_EXECUTABLE=$(which python)```?

akshaya-code, 2021-11-18:
Which directory should be the "your_install_path"