limbo018/DREAMPlace #48: CMAKE build issues

ayeung-fw, 2021-09-04:
The following step did not work for me (due to CMake version mismatch 3.8.2 vs. 3.14):
docker pull limbo018/dreamplace:cuda
[image]

As a result, I had to rebuild docker instead:
docker build . --file Dockerfile --tag ayeung/dreamplace:cuda

Then I hit another issue as I had to modify CMakeLists.txt to remove CUDA_VERSION_MAJOR version check:
original
[image]
new
[image]

Thanks,
Adrian

limbo018, 2021-09-05:
Hi Adrian, thanks for the feedback. I have just pushed the new docker image built from the latest dockerfile. 

On my side, it looks like there is no need to modify the CMakeLists.txt. I tried to build with the docker image and it works fine. Is it because your local cuda version is not CUDA 11?

Thanks
Yibo

ayeung-fw, 2021-09-06:
I checked the CUDA version number and it was 11, which it why I disabled the version check.

And I believe another user at my company also had a similar problem.

thesukantadey, 2021-11-18:
I have also received similar error when executed the following command (without docker)....
`$ cmake .. -DCMAKE_INSTALL_PREFIX=/home/sukanta/install/ -DPYTHON_EXECUTABLE=$(which python)`
and got the following:

-- CMAKE_INSTALL_PREFIX: /home/sukanta/install
-- CMAKE_BUILD_TYPE: Release
-- CMAKE_CXX_ABI: _GLIBCXX_USE_CXX11_ABI=0
-- pybind11 v2.6.3 dev1
-- TORCH_INSTALL_PREFIX=/home/sukanta/.local/lib/python3.8/site-packages/torch
-- TORCH_VERSION=1.10
-- TORCH_ENABLE_CUDA=0
-- Boost version: 1.72.0
-- Boost_INCLUDE_DIRS = /usr/local/include
-- Cairo: /usr/include/cairo
-- Cairo: /usr/lib/x86_64-linux-gnu/libcairo.so
-- OPS_DIR /home/sukanta/DREAMPlace/dreamplace/ops
-- UTILITY_LIBRARY_DIRS /home/sukanta/DREAMPlace/build/dreamplace/ops/utility
-- FLUTE_INCLUDE_DIRS /home/sukanta/DREAMPlace/thirdparty/flute-3.1
-- FLUTE_LINK_DIRS /home/sukanta/DREAMPlace/build/thirdparty/flute-3.1
-- LIMBO_SOURCE_DIR /home/sukanta/DREAMPlace/thirdparty/Limbo
-- LIMBO_BINARY_DIR /home/sukanta/DREAMPlace/build/thirdparty/Limbo
CMake Error at CMakeLists.txt:85 (if):
  if given arguments:

    "CUDA_FOUND" "AND" "VERSION_GREATER_EQUAL" "11"

  Unknown arguments specified

-- Configuring incomplete, errors occurred!
See also "/home/sukanta/DREAMPlace/build/CMakeFiles/CMakeOutput.log".
CMake Error: Unable to open check cache file for write. /home/sukanta/DREAMPlace/build/CMakeFiles/cmake.check_cache

Please kindly help.

akshaya-code, 2021-11-21:
I only changed things as mentioned  #48 (comment) and able to build it with some warnings.However when I try to run using the following command
python dreamplace/Placer.py test/ispd2005/adaptec1.json it failed with the following message though it contains other things before this message

Traceback (most recent call last):
  File "dreamplace/Placer.py", line 174, in <module>
    place(params)
  File "dreamplace/Placer.py", line 32, in place
    assert (not params.gpu) or configure.compile_configurations["CUDA_FOUND"] == 'TRUE', \
AssertionError: CANNOT enable GPU without CUDA compiled

Kindly look into it

thesukantadey, 2021-11-21:
That's not an error. Disable your GPU by setting GPU flag to 0 in the adaptec1.json file. It will work.

akshaya-code, 2021-11-22:
Thank You brother it worked.