limbo018/DREAMPlace #50: Parser Def meet errors

Yangxiaojun1230, 2021-09-10:
Hi author,
Thanks for your work, it's a great project! 
I am a new guy for this domain. Now I am trying to read in my own def files which were already finished placement, but meet some error , could you give me some help?

```
END VIAS
- VIA67_1cut_BW38_UW38_ISO_17_2_2  ;
+ VIARULE 'VIA67_1cut_BW38_UW38_ISO'
  + CUTSIZE 76 76
  + LAYERS M6 VIA6 M7
  + CUTSPACING 148 148
  + ENCLOSURE 130 0 0 130
  + ROWCOL 17 2
END VIAS
- VIA78_1cut_1_4_2  ;
+ VIARULE 'VIA78_1cut'
  + CUTSIZE 648 648
  + LAYERS M7 VIA7 M8
  + CUTSPACING 612 612
  + ENCLOSURE 36 144 144 36
  + ROWCOL 1 4
END VIAS
- VIA89_1cut_1_5_2  ;
+ VIARULE 'VIA89_1cut'
  + CUTSIZE 648 648
  + LAYERS M8 VIA8 M9
  + CUTSPACING 612 612
  + ENCLOSURE 144 36 36 144
  + ROWCOL 1 5
END VIAS

COMPONENTMASKSHIFT M3 VIA2 M2 VIA1 M1 M0 ;
---------------------------------------------------------------------------
IndexError                                Traceback (most recent call last)
<timed eval> in <module>

/DREAMPlace/DREAMPlace/dreamplace/PlaceDB.py in __call__(self, params)
    550         tt = time.time()
    551 
--> 552         self.read(params)
    553         self.initialize(params)
    554 

/DREAMPlace/DREAMPlace/dreamplace/PlaceDB.py in read(self, params)
    433         """
    434         self.dtype = datatypes[params.dtype]
--> 435         self.rawdb = place_io.PlaceIOFunction.read(params)
    436         self.initialize_from_rawdb(params)
    437 

/DREAMPlace/DREAMPlace/dreamplace/ops/place_io/place_io.py in read(params)
     33             args += " --sort_nets_by_degree %s" % (params.sort_nets_by_degree)
     34 
---> 35         return place_io_cpp.forward(args.split(' '))
     36 
     37     @staticmethod

IndexError: vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
```

limbo018, 2021-09-10:
It looks like something went wrong in the DEF parser or the database. You can try building the code in Debug mode and use gdb to locate which line of code goes wrong. In this way, you can derive which line in the DEF file is incompatible with the parser or the database.

Yangxiaojun1230, 2021-09-11:
> It looks like something went wrong in the DEF parser or the database. You can try building the code in Debug mode and use gdb to locate which line of code goes wrong. In this way, you can derive which line in the DEF file is incompatible with the parser or the database.

Thanks for your reply.
I located the error in "thirdparty/Limbo/limbo/parsers/def/adapt/DefDriver.cc"   at line 2653
When I want to step into the function, the function run away. I have no idea about it. Do you have any suggestion?
I checked the def file version is 5.7 and the file was wrote out from ICC2 after global routing.
```
(gdb) c
Continuing.

Thread 1.1 "python" hit Breakpoint 5, DefParser::Driver::parse_file (this=0x7fffffffcbd0, 
    filename=...) at /DREAMPlace/thirdparty/Limbo/limbo/parsers/def/adapt/DefDriver.cc:2653
2653	    int res = defrRead(f, filename.c_str(), userData, 1);
(gdb) s
std::string::c_str (this=0x7fffffffd468)
    at /home/nwani/m3/conda-bld/compilers_linux-64_1560109574129/work/.build/x86_64-conda_cos6-linux-gnu/build/build-cc-gcc-final/x86_64-conda_cos6-linux-gnu/libstdc++-v3/include/bits/basic_string.h:5195
5195	/home/nwani/m3/conda-bld/compilers_linux-64_1560109574129/work/.build/x86_64-conda_cos6-linux-gnu/build/build-cc-gcc-final/x86_64-conda_cos6-linux-gnu/libstdc++-v3/include/bits/basic_string.h: No such file or directory.
(gdb) n
.....
.....
  + ENCLOSURE 144 36 36 144
  + ROWCOL 1 5
END VIAS
0x00007fff18db0ca9 in DefParser::read (db=..., defFile=...)
    at /DREAMPlace/thirdparty/Limbo/limbo/parsers/def/adapt/DefDriver.cc:2786
2786	}
(gdb) n
2781		Driver driver (db);
(gdb) 
0x00007fff18d29568 in DreamPlace::place_io_forward (args=...)
    at /DREAMPlace/dreamplace/ops/place_io/src/place_io.cpp:113
113	        std::string token = pybind11::str(args[i]);
```

Yangxiaojun1230, 2021-09-13:
UPdate debug process:
Now I could step into function defrRead() , and then step into defyyparse(defData) in file "thirdparty/Limbo/limbo/thirdparty/lefdef/5.8/def/def/defrReader.cpp" . And yyparse defined as defyyparse in file "build/thirdparty/Limbo/limbo/thirdparty/lefdef/5.8/def/def.tab.cpp". When I go through the def.tab.cpp , I found  a problem in below function. In each case, "def/def.y" throw the exception "No such file". Actually I could find the def.y file in path "thirdparty/Limbo/limbo/thirdparty/lefdef/5.8/def/def/def.tab.cpp", I guess there is some links should be established when building. 
```
switch (yyn)
    {
  case 4:
#line 220 "def/def.y"
    { defData->dumb_mode = 1; defData->no_num = 1; }
#line 2929 "/DREAMPlace/build/thirdparty/Limbo/limbo/thirdparty/lefdef/5.8/def/def.tab.cpp"
    break;

  case 5:
#line 221 "def/def.y"
    {
        defData->VersionNum = defrData::con
```

limbo018, 2021-09-13:
Hi, this may not be an easy way to find the problem. From the information you provided so far, I believe the problem lies incompatible syntax in the DEF file with the parser. 

One alternative way to debug is "binary search" the input DEF file. You can gradually remove some contents in the DEF file until you see different error reported or it does not report any error. This can help you narrow down the range in the DEF file.

Yangxiaojun1230, 2021-09-13:
> Hi, this may not be an easy way to find the problem. From the information you provided so far, I believe the problem lies incompatible syntax in the DEF file with the parser.
>
> One alternative way to [...]

Thanks for your reply.
And I tried just like you said, and finally found the root of the problem. The reason is when using command write_lef in ICC2, it need to add the option -design, otherwise, the lef file would miss some information that lead errors when read def.

closed 2021-09-13 by Yangxiaojun1230