limbo018/DREAMPlace #54: How to place the DAC-2012 benchmark suite

KaitoKiddo, 2021-12-21:
I added dac-2012 benchmark suite in ./benchmarks, then I run the cmd "python dreamplace/Placer.py test/dac2012/superblue2.json".
It could place the superblue2 benchmark and generated the .pl file "superblue2.NCTUgr.pl". 
But when reading the NCTU-GR 2.0, it threw an error: 
FileNotFoundError: [Errno 2] No such file or directory: '...../results/superblue2/superblue2.NCTUgr.ofinfo'.
Why is there no such file in ./results ?

limbo018, 2021-12-28:
> Hi, I'm sorry to bother. I could place the ispd2005 benchmark suite, but when I tried to place DAC-2012 benchmark suite I met some problems.
> 
> I added dac-2012 benchmark suite in ./benchmarks, then I run the cmd "python dreamplace/Placer.py test/dac2012/superblue2.json".
> 
> It could place the superblue2 benchmark and generated the .pl file "superblue2.NCTUgr.pl".
> 
> But when reading the NCTU-GR 2.0, it threw an error: FileNotFoundError: [Errno 2] No such file or directory: '...../results/superblue2/superblue2.NCTUgr.ofinfo'. Why is there no such file in ./results ?

Hi, I just tried running superblue2 in DAC 2012 benchmarks. It works fine on my side. The *.ofinfo file is generated by NCTUgr under results/superblue2/ of your install directory. 

Could you check if NCTUgr crashed on your machine?

Thanks

KaitoKiddo, 2021-12-29:
Thank you for your reply. 
In thirdparty, the NCTUgr.ICCAD2012 folder contains NCTUgr, DAC12.set, ICCAD12.set files, etc.
Does the NCTUgr need to be downloaded from DAC2012 contest or can be deployed to my machine when cmake?

limbo018, 2021-12-29:
You can use the NCTUgr binary in the folder. I doubt the binary probably crashed when running on your machine. I actually downloaded the binary from the contest website.

Eternity666, 2022-03-17:
Hi, sorry to bother! I met the same problem as you. Have you solved it?

limbo018, 2022-03-17:
Not sure if this can help. My machine is ubuntu 18.04 LTS with NVIDIA 2080TI GPU. You'd better make sure NCTUgr does not crash on standalone runs, as DREAMPlace will call the NCTUgr binary during the optimization.

Eternity666, 2022-03-17:
Running NCTUgr will result in a segmentation fault. When I view the core file through GDB, this information will be displayed. Do you know how to solve it? Thank you!

[image]

limbo018, 2022-03-17:
It looks like the binary does not fit your machine. NCTUgr is not open-source, so unfortunately I don't know how it was written either.

gjhhust, 2023-03-18:
> Running NCTUgr will result in a segmentation fault. When I view the core file through GDB, this information will be displayed. Do you know how to solve it? Thank you!
> 
> [image]

I also have same error，Did you solve?

gjhhust, 2023-03-18:
> > > Hi, I'm sorry to bother. I could place the ispd2005 benchmark suite, but when I tried to place DAC-2012 benchmark suite I met some problems.
> > > I added dac-2012 benchmark suite in ./benchmarks, then I run the cmd "python dreamplace/Placer.py test/dac2012/superblue2.json".
> > > It could place the superblue2 benchmark and generated the .pl file "superblue2.NCTUgr.pl".
> > > But when reading the NCTU-GR 2.0, it threw an error: FileNotFoundError: [Errno 2] No such file or directory: '...../results/superblue2/superblue2.NCTUgr.ofinfo'. Why is there no such file in ./results ?
> > 
> > 
> > Hi, I just tried running superblue2 in DAC 2012 benchmarks. It works fine on my side. The *.ofinfo file is generated by NCTUgr under results/superblue2/ of your install directory.
> > Could you check if NCTUgr crashed on your machine?
> > Thanks
> 
> Thank you for your reply. In thirdparty, the NCTUgr.ICCAD2012 folder contains NCTUgr, DAC12.set, ICCAD12.set files, etc. Does the NCTUgr need to be downloaded from DAC2012 contest or can be deployed to my machine when cmake?

I'm sorry to bother you, but did you solve it? I had the same problem

ZhuLinsen, 2023-06-25:
@KaitoKiddo @Eternity666  @gjhhust hello, I'm sorry to bother you, but did you solve it? I had the same problem. 
I'm puzzled as to why it is throwing errors because I am running it in a Docker environment.

Any suggestions would be greatly appreciated.Thank you.

KaitoKiddo, 2023-06-25:
I have finished placing dac2012 benchmark suite, without error. But I'm sorry that I don't run it in docker. @gjhhust @ZhuLinsen

gjhhust, 2023-06-25:
> 

I was using superblue1 to run fast and default, but there was no problem with dac2012 benchmark. To see if NCTUGR is the problem, try NCTUGR on its own: https://[people.cs.nctu.edu.tw/~whliu/NCTU-GR.htm](https://people.cs.nctu.edu.tw/~whliu/NCTU-GR.htm)

gjhhust, 2023-06-25:
> I have finished placing dac2012 benchmark suite, without error. But I'm sorry that I don't run it in docker. @gjhhust @ZhuLinsen

Sorry, I used to run in my environment, and this issue has my environment description: https://github.com/limbo018/DREAMPlace/issues/113

KaitoKiddo, 2023-06-25:
I have never met the errors that you mentioned. In DREAMPlace it has the NCTUgr and will be run when placing dac2012 benchmark. Maybe you can check your environment or docker. Thank you for your attention to this issue. Closing it.

closed 2023-06-25 by KaitoKiddo

ZhuLinsen, 2023-06-25:
> 
Hi @KaitoKiddo @gjhhust, Thank you for your response, so how did you solve the error as below？
`FileNotFoundError: [Errno 2] No such file or directory: '...../results/superblue2/superblue2.NCTUgr.ofinfo'.
Why is there no such file in ./results ?`

KaitoKiddo, 2023-06-25:
Maybe you need to check your NCTUgr that is working.

xiaoqiang88888, 2025-03-15:
@KaitoKiddo @limbo018 @ZhuLinsen @Eternity666 @gjhhust Respected teachers, I sincerely apologize for the interruption. Could you please advise on how to resolve this issue? I have tried many times, but the NCTUgr keeps being broken.

[image]

KaitoKiddo, 2025-03-17:
Hi, you should check to see how the NCTUgr tool works I think, and this is not a problem within DREAMPlace.

xiaoqiang88888, 2025-03-19:
@KaitoKiddo 老师您好，我觉得我的NCTUgr应该是work的如下图所示，能正常的routing，也生成相应的文件，但是唯独没有生成ofinfo文件，方便加您一个联系方式么，如果能解决将不胜感激 qq：903497079

[image]

[image]