limbo018/DREAMPlace #57: ould NOT find LpSolve (missing: LPSOLVE_LIBRARIES LPSOLVE_INCLUDE_DIRS LPSOLVE_LINKS)

LiuX41, 2021-12-28:
I'm building with docker, but when commanding
`cmake .. -DCMAKE_INSTALL_PREFIX=your_install_path -DPYTHON_EXECUTABLE=$(which python)`
in the docker bash, the error occurs:
-- Could NOT find LpSolve (missing: LPSOLVE_LIBRARIES LPSOLVE_INCLUDE_DIRS LPSOLVE_LINKS)
-- Could NOT find Doxygen (missing: DOXYGEN_EXECUTABLE) 
-- Could NOT find Ghostscript (missing: GHOSTSCRIPT_EXECUTABLE) 
-- Could NOT find COIN (missing: COIN_INCLUDE_DIR COIN_CBC_LIBRARY COIN_CBC_SOLVER_LIBRARY COIN_CGL_LIBRARY COIN_CLP_LIBRARY COIN_COIN_UTILS_LIBRARY COIN_OSI_LIBRARY COIN_OSI_CBC_LIBRARY COIN_OSI_CLP_LIBRARY) 

Actually, when I was trying to build without docker before, the same problem had occurred.

Thank you!

limbo018, 2022-01-26:
Is your CMake process complete? Missing LpSolve only generates warning.

PandaK404, 2022-10-13:
Hi, bro, I encountered the same problem. How did you solve it,