limbo018/DREAMPlace #58: During following error occured. Pleaseelp if any body knows it

bnbray, 2022-01-01:
bnb@BNB-HP:~/DREAMPlace$ python dreamplace/Placer.py test/ispd2005/adaptec1.json
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 21, in <module>
    import PlaceDB
  File "/home/bnb/DREAMPlace/dreamplace/PlaceDB.py", line 16, in <module>
    import dreamplace.ops.place_io.place_io as place_io 
  File "/home/bnb/DREAMPlace/dreamplace/ops/place_io/place_io.py", line 9, in <module>
    import dreamplace.ops.place_io.place_io_cpp as place_io_cpp
ImportError: /home/bnb/DREAMPlace/dreamplace/ops/place_io/place_io_cpp.cpython-35m-x86_64-linux-gnu.so: undefined symbol: _ZN3c106Symbol14fromQualStringERKSs

liumengbjut, 2022-01-25:
It seems you did not run the code in you installed folder.

bnbray, 2022-01-25:
Thanks for the reply

closed 2022-01-25 by bnbray