limbo018/DREAMPlace #60: Issue on making: nvcc fatal   : Unsupported gpu architecture 'compute_86'

Dinple, 2022-01-13:
Hi all,

I got DREAMPlace running successfully on my computer using the CPU version. However, when trying to make using the nvidia docker, I kept on getting the error saying nvcc fatal   : Unsupported gpu architecture 'compute_86'.

The error is shown below:
[image]

I followed the installation guide from the nvidia docker repository and even tested the installation. Here is nvidia-smi and nvcc -V inside docker container:
[image]
[image]

Since the original system I was using is built on a 3090, after I did some search online, some suggested a compatibility issue with cuda 11 and 30-series card. However, after redoing the whole installation on a gtx 1080 with cuda 11, same issue occurs again. 

I am not really sure if there is a solution to this issue but if anyone is running DREAMPlace on a GPU, could you let me know which CUDA version and GPU you are using?

limbo018, 2022-01-13:
Hi, if you switch to GTX 1080, compute_86 does not work. You need to check the list and set the correct architecture compatibility (it is mentioned in the README). 

https://developer.nvidia.com/cuda-gpus

Thanks

Dinple, 2022-01-17:
Hi, thanks for the info. I eventually solved the issue by simply change the pytorch version to 1.9.0 and cuda version >= 11.1 in the dockerfile and rebuild the container.

closed 2022-01-17 by Dinple