limbo018/DREAMPlace #61: DRieamPlace  cmake issue

bnbray, 2022-01-25:
cmake .. -DCMAKE_INSTALL_PREFIX=~/newdream/DREAMPlace/instal/  -DPYTHON_EXECUTABLE=3.9.5
CMake Error at /usr/local/share/cmake-3.20/Modules/CMakeDetermineCUDACompiler.cmake:25 (message):
  Could not find compiler set in environment variable CUDACXX:

  /usr/local/cuda/bin/nvcc.

Call Stack (most recent call first):
  CMakeLists.txt:3 (project)

-- Configuring incomplete, errors occurred!
See also "/home/bnb/newdream/DREAMPlace/build/CMakeFiles/CMakeOutput.log".
See also "/home/bnb/newdream/DREAMPlace/build/CMakeFiles/CMakeError.log"

limbo018, 2022-01-26:
Hi, do you want to install it with GPU support or CPU only?

bnbray, 2022-01-30:
Only CPU support not GPU