limbo018/DREAMPlace #62: DRieam Place compilation problem in CPU only

bnbray, 2022-02-05:
(base) bnb@BNB-HP:~/newdream/DREAMPlace/build$ cmake .. -DCMAKE_INSTALL_PREFIX= ~/DREAMPlace/install -DPYTHON_EXECUTABLE=3.9
-- The CUDA compiler identification is NVIDIA 9.1.85
-- Detecting CUDA compiler ABI info
-- Detecting CUDA compiler ABI info - done
-- Check for working CUDA compiler: /usr/bin/nvcc - skipped
-- Detecting CUDA compile features
-- Detecting CUDA compile features - done
-- CMAKE_INSTALL_PREFIX: 
-- CMAKE_BUILD_TYPE: Release
-- CMAKE_CXX_ABI: _GLIBCXX_USE_CXX11_ABI=0
-- pybind11 v2.6.3 dev1
-- Found PythonInterp: /home/bnb/newdream/DREAMPlace/build/3.9  
CMake Error at thirdparty/pybind11/tools/FindPythonLibsNew.cmake:133 (message):
  Python config failure:

Call Stack (most recent call first):
  thirdparty/pybind11/tools/pybind11Tools.cmake:45 (find_package)
  thirdparty/pybind11/tools/pybind11Common.cmake:205 (include)
  thirdparty/pybind11/CMakeLists.txt:190 (include)

-- Configuring incomplete, errors occurred!
See also "/home/bnb/newdream/DREAMPlace/build/CMakeFiles/CMakeOutput.log".
See also "/home/bnb/newdream/DREAMPlace/build/CMakeFiles/CMakeError.log".

limbo018, 2022-02-05:
This ```-DPYTHON_EXECUTABLE=3.9``` looks weird. You can omit it and have a try. Or you can replace 3.9 with a full path to the python executable.

bnbray, 2022-02-07:
thanks for the correction

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>