limbo018/DREAMPlace #67: Could NOT find LpSolve and COIN

WOOSHIK-M, 2022-03-26:
Hi, I want to try running your DREAMPlace.
It seems very awesome!

However, I got the below message when implementing `cmake ..`.
```
-- CMAKE_INSTALL_PREFIX: /home/wooshik.myung/MRX/DREAMPlace/install
-- CMAKE_BUILD_TYPE: Release
-- CMAKE_CXX_ABI: _GLIBCXX_USE_CXX11_ABI=0
-- pybind11 v2.6.3 dev1
-- TORCH_INSTALL_PREFIX=/home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch
-- TORCH_VERSION=1.11
-- TORCH_ENABLE_CUDA=1
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.55.0")
-- Boost_INCLUDE_DIRS = /usr/include
-- Cairo: /usr/include/cairo
-- Cairo: /usr/lib/x86_64-linux-gnu/libcairo.so
-- OPS_DIR /home/wooshik.myung/MRX/DREAMPlace/dreamplace/ops
-- UTILITY_LIBRARY_DIRS /home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/ops/utility
-- FLUTE_INCLUDE_DIRS /home/wooshik.myung/MRX/DREAMPlace/thirdparty/flute-3.1
-- FLUTE_LINK_DIRS /home/wooshik.myung/MRX/DREAMPlace/install/thirdparty/flute-3.1
-- LIMBO_SOURCE_DIR /home/wooshik.myung/MRX/DREAMPlace/thirdparty/Limbo
-- LIMBO_BINARY_DIR /home/wooshik.myung/MRX/DREAMPlace/install/thirdparty/Limbo
-- CUB_DIR /usr/local/cuda-11.2/include
-- MUNKRES_CPP_INCLUDE_DIRS /home/wooshik.myung/MRX/DREAMPlace/thirdparty/munkres-cpp/src
-- MUNKRES_CPP_LINK_DIRS /home/wooshik.myung/MRX/DREAMPlace/install/thirdparty/munkres-cpp
-- LEMON_INCLUDE_DIRS /home/wooshik.myung/MRX/DREAMPlace/thirdparty/Limbo/limbo/thirdparty/lemon;/home/wooshik.myung/MRX/DREAMPlace/install/thirdparty/Limbo/limbo/thirdparty/lemon
-- LEMON_LINK_DIRS /home/wooshik.myung/MRX/DREAMPlace/install/thirdparty/Limbo/limbo/thirdparty/lemon/lemon
-- CUDA_ARCH_FLAGS: -gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86
-- CMAKE_BUILD_TYPE: Release
-- CMAKE_CXX_ABI: _GLIBCXX_USE_CXX11_ABI=0
-- OPENBLAS: OFF
-- INSTALL_LIMBO: OFF
-- GENERATE_DOCS: OFF
-- ENABLE_TEST: OFF
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.55.0") found components: graph regex
-- Could not link against lpsolve55!
-- Could NOT find LpSolve (missing: LPSOLVE_LIBRARIES LPSOLVE_LINKS)
CMake Deprecation Warning at thirdparty/Limbo/limbo/thirdparty/lemon/CMakeLists.txt:4 (CMAKE_POLICY):
  The OLD behavior for policy CMP0048 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Could NOT find COIN (missing: COIN_INCLUDE_DIR COIN_CBC_LIBRARY COIN_CBC_SOLVER_LIBRARY COIN_CGL_LIBRARY COIN_CLP_LIBRARY COIN_COIN_UTILS_LIBRARY COIN_OSI_LIBRARY COIN_OSI_CBC_LIBRARY COIN_OSI_CLP_LIBRARY)
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wooshik.myung/MRX/DREAMPlace/install
```
I installed `lp_solve` using `sudo apt-get install lp_solve` and also checked it is installed at `/usr/bin` normally.

I tried `cmake .. -DLPSOLVE_INCLUDE_PATH="/usr/bin/lp_solve"`, but it still occur the same error.

How to solve it? 
Thanks!

limbo018, 2022-03-26:
Hi, this is just a warning from a thirdparty library. I don't think you need to worry about it.

WOOSHIK-M, 2022-03-26:
Thanks for your reply.
Does this message cause any other issues when running DREAMPlace?

limbo018, 2022-03-26:
You can have a try. I don't remember any issue without lpsolve.

WOOSHIK-M, 2022-03-26:
What about `COIN`? Is it also unnecessary?

WOOSHIK-M, 2022-03-26:
I got the below message, after running `python dreamplace/Placer.py test/ispd2005/adaptec1.json`.

```
Traceback (most recent call last):
  File "dreamplace/Placer.py", line 22, in <module>
    import NonLinearPlace
  File "/home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/NonLinearPlace.py", line 23, in <module>
    import BasicPlace
  File "/home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/BasicPlace.py", line 32, in <module>
    import dreamplace.ops.independent_set_matching.independent_set_matching as independent_set_matching
  File "/home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/ops/independent_set_matching/independent_set_matching.py", line 17, in <module>
    import dreamplace.ops.independent_set_matching.independent_set_matching_cuda as independent_set_matching_cuda
ImportError: /home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/ops/independent_set_matching/independent_set_matching_cuda.cpython-38-x86_64-linux-gnu.so: undefined symbol: curandCreateGenerator
```

Is this caused by CUDA?
How to fix it?
 
Thanks!

limbo018, 2022-03-26:
Yes. Please first check the following section carefully in the readme. It is highly related to your CUDA version and GPU card. 

[image]

WOOSHIK-M, 2022-03-26:
Thanks for your reply!!

I found my GPU setting as below pictures.

```
CUDA: 11.4
Compute Capability: 8.0
```
[image]
[image]

I see your kind description, but there is only until 7.5.
Does it work if I set the `CMAKE_CUDA_FLAGS` to `gencode=arch=compute_70, code=sm_70`?
if it works, when do I add this option? `make install`?

Thanks!!!

limbo018, 2022-03-26:
I saw the cmake log contains 

```
-- CUDA_ARCH_FLAGS: -gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86
```

So compatibility 8.0 should already be covered. 

It looks like curand is not linked when compiling the independent_set_matching op. Can you check by ```male clean; make install VERBOSE=1 >& log``` ? You can check the make command in the log for building the independent_set_matching op. You can see whether curand lib is linked (the lib comes with CUDA).

WOOSHIK-M, 2022-03-26:
Thank you for your reply.
It didn't react when running `make install VERBOSE=1 >& log`, thus I removed the current repo and re-installed.

I saw the below message about building `independent_set_matching_cuda.cpython-38-x86_64-linux-gnu.so`, but didn't find any errors occur.

```
[ 91%] Linking CXX shared module independent_set_matching_cuda.cpython-38-x86_64-linux-gnu.so
cd /home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/ops/independent_set_matching && /usr/local/bin/cmake -E cmake_link_script CMakeFiles/independent_set_matching_cuda.dir/link.txt --verbose=1
/usr/bin/c++ -fPIC -O3 -DNDEBUG -flto -shared  -o independent_set_matching_cuda.cpython-38-x86_64-linux-gnu.so CMakeFiles/independent_set_matching_cuda.dir/src/independent_set_matching_cuda.cpp.o  -Wl,-rpath,/home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib:/usr/local/cuda-11.2/lib64: libindependent_set_matching_cuda_cuda_tmp.a ../utility/libutility_cxx.a ../../../thirdparty/Limbo/limbo/parsers/gdsii/stream/libgdsparser.a ../../../thirdparty/Limbo/limbo/thirdparty/gzstream/libgzstream.a ../../../thirdparty/munkres-cpp/libmunkres.a ../../../thirdparty/Limbo/limbo/thirdparty/lemon/lemon/libemon.a /usr/lib/x86_64-linux-gnu/libz.so /usr/lib/x86_64-linux-gnu/libcairo.so /home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib/libtorch.so /home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib/libtorch_python.so /home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib/libc10.so /home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib/libtorch_cpu.so /home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib/libc10_cuda.so /home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/lib/libtorch_cuda.so /usr/lib/gcc/x86_64-linux-gnu/8/libgomp.so -lpthread /usr/local/cuda-11.2/lib64/libcudart.so /usr/local/cuda-11.2/lib64/libcudadevrt.a ../../../thirdparty/Limbo/limbo/parsers/gdsii/stream/libgdsparser.a ../../../thirdparty/Limbo/limbo/thirdparty/gzstream/libgzstream.a /usr/lib/x86_64-linux-gnu/libz.so /usr/lib/x86_64-linux-gnu/libcairo.so
cd /home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/ops/independent_set_matching && /usr/bin/strip /home/wooshik.myung/MRX/DREAMPlace/install/dreamplace/ops/independent_set_matching/independent_set_matching_cuda.cpython-38-x86_64-linux-gnu.so
make[2]: Leaving directory '/home/wooshik.myung/MRX/DREAMPlace/install'
[ 91%] Built target independent_set_matching_cuda
make  -f dreamplace/ops/k_reorder/CMakeFiles/k_reorder_cuda_cuda_tmp.dir/build.make dreamplace/ops/k_reorder/CMakeFiles/k_reorder_cuda_cuda_tmp.dir/depend
make[2]: Entering directory '/home/wooshik.myung/MRX/DREAMPlace/install'
```

Is there any unnormal message?

limbo018, 2022-03-26:
Hi, I found the problem, but we need to try a little bit to verify the solution, as I cannot reproduce your problem on my side. Can you try modifying the line 54 of the file dreamplace/ops/independent_set_matching/CMakeLists.txt. 
From 
```EXTRA_LINK_LIBRARIES ${LINK_LIBS}```
To 
```EXTRA_LINK_LIBRARIES ${LINK_LIBS} ${CUDA_curand_LIBRARY}```

Then, you can try rerun ```make install``` in the build directory. 

Let me know if it works.

limbo018, 2022-03-26:
I think the problem is your pytorch installation now. 

```
/home/wooshik.myung/anaconda3/envs/DREAMPlace/lib/python3.8/site-packages/torch/cuda/__init__.py:145: UserWarning:
NVIDIA A100-PCIE-40GB with CUDA capability sm_80 is not compatible with the current PyTorch installation.
The current PyTorch install supports CUDA capabilities sm_37 sm_50 sm_60 sm_70
```

The kernel image error usually comes from incompatible GPU architecture.

WOOSHIK-M, 2022-03-26:
OMG!
Looks like it is almost there.
I go to find out how to fix it. Thanks!

WOOSHIK-M, 2022-03-27:
I ran this finally! I really appreciate your help. 
I fixed this issue from the below command.
```
pip3 install torch==1.9.1+cu111 torchvision==0.10.1+cu111 torchaudio==0.9.1 -f https://download.pytorch.org/whl/torch_stable.html
```

I have the last question.
Are there any `docs` about the setting of DREAMPlace? e.g. Some parameters and netlist file format...etc.

limbo018, 2022-03-27:
Hi, DREAMPlace supports bookshelf format (https://vlsicad.ucsd.edu/GSRC/bookshelf/Slots/Placement/plFormats.html) and lef/def format (https://www.ispd.cc/contests/18/lefdefref.pdf). 

There is no detailed docs for now. For the parameters in the JSON configuration, it is defined in the README. You may need to read the DREAMPlace paper and the ePlace paper to understand the fundamental mechanism of the algorithm.

WOOSHIK-M, 2022-03-27:
I really appreciate your help.
I close this issue. Thanks!!

closed 2022-03-27 by WOOSHIK-M