limbo018/DREAMPlace #68: Bookshelf file parsing failed

Eternity666, 2022-03-27:
Sorry to bother. When I run ```python ./dreamplace/Placer.py test/dac2012/superblue2.json``` in the install_path, an error occured. 
[image]

Would you please advise how to solve it? Thank you!

limbo018, 2022-03-27:
It looks like DAC 2012 benchmarks are missing. Have you downloaded them and put in the correct locations as the json file defined?

Eternity666, 2022-03-30:
Thanks for your reply! I solved this problem！

closed 2022-04-02 by limbo018

PKUTAN, 2022-04-15:
> Thanks for your reply! I solved this problem！

How did you solve it exactly?

Eternity666, 2022-04-16:
Just put the benchmark data in the correct path. Alternatively, you can modify the path to the correct path of the .aux file in corresponding json file.

PKUTAN, 2022-04-16:
> Just put the benchmark data in the correct path. Alternatively, you can modify the path to the correct path of the .aux file in corresponding json file.

That's wield for my case, I have put the file in the correct path, however the error still occur.