limbo018/DREAMPlace #69: LEF/DEF Format error

WOOSHIK-M, 2022-03-28:
Hi, Prof. Lin.

I really appreciate your help with the installation of our environment.
However, I still have a problem when running DREAMPlace with LEF/DEF file format.
I extracted LEF/DEF format file after macro placement of our chip design from [IC Compiler II](https://www.synopsys.com/content/dam/synopsys/implementation&signoff/datasheets/ic-compiler-ii-ds.pdf) Tool.

But the result looks unnormal.
The process is terminated with the below error.

```
========================================================
                       DREAMPlace
            Yibo Lin (http://yibolin.com)
   David Z. Pan (http://users.ece.utexas.edu/~dpan)
========================================================
[INFO   ] DREAMPlace - parameters = {'aux_input': '', 'lef_input': 'design.lef', 'def_input': 'design.def', 'verilog_input': '', 'gpu': 1, 'num_bins_x': 512, 'num_bins_y': 512, 'global_place_stages': [{'num_bins_x': 512, 'num_bins_y': 512, 'iteration': 1000, 'learning_rate': 0.01, 'wirelength': 'weighted_average', 'optimizer': 'nesterov', 'Llambda_density_weight_iteration': 1, 'Lsub_iteration': 1}], 'target_density': 1.0, 'density_weight': 8e-05, 'random_seed': 1000, 'result_dir': 'results', 'scale_factor': 1.0, 'ignore_net_degree': 100, 'gp_noise_ratio': 0.025, 'enable_fillers': 1, 'global_place_flag': 1, 'legalize_flag': 1, 'detailed_place_flag': 1, 'stop_overflow': 0.07, 'dtype': 'float32', 'detailed_place_engine': '', 'detailed_place_command': '', 'plot_flag': 0, 'RePlAce_ref_hpwl': 350000, 'RePlAce_LOWER_PCOF': 0.95, 'RePlAce_UPPER_PCOF': 1.05, 'gamma': 4.0, 'RePlAce_skip_energy_flag': 0, 'random_center_init_flag': 1, 'sort_nets_by_degree': 0, 'num_threads': 8, 'dump_global_place_solution_flag': 0, 'dump_legalize_solution_flag': 0, 'routability_opt_flag': 0, 'route_num_bins_x': 512, 'route_num_bins_y': 512, 'node_area_adjust_overflow': 0.15, 'max_num_area_adjust': 3, 'adjust_nctugr_area_flag': 0, 'adjust_rudy_area_flag': 1, 'adjust_pin_area_flag': 1, 'area_adjust_stop_ratio': 0.01, 'route_area_adjust_stop_ratio': 0.01, 'pin_area_adjust_stop_ratio': 0.05, 'unit_horizontal_capacity': 1.5625, 'unit_vertical_capacity': 1.45, 'unit_pin_capacity': 0.058, 'max_route_opt_adjust_rate': 2.0, 'route_opt_adjust_exponent': 2.0, 'pin_stretch_ratio': 1.414213562, 'max_pin_opt_adjust_rate': 1.5, 'deterministic_flag': 0}

...

[WARNING] unsupported shape 3 in LEF file
[WARNING] unsupported shape 3 in LEF file
[WARNING] unsupported shape 3 in LEF file
[WARNING] unsupported shape 3 in LEF file
[WARNING] unsupported shape 3 in LEF file
[WARNING] unsupported shape 3 in LEF file

...

[INFO   ] reading design.def
[INFO   ] detect 1069 rows, 633339 components, 1370 IO pins, 638286 nets, 209 blockages
[WARNING] no layer specified by pin VDDI
[WARNING] no position or bounding box specified by pin VDDI, set to (0, 0, 0, 0)
[WARNING] no layer specified by pin VSSI
[WARNING] no position or bounding box specified by pin VSSI, set to (0, 0, 0, 0)
- PIN clk_i ;
  + PROPERTY ACCESS_DIRECTION 0 top STRING ;
END PINPROPERTIES
- PIN nsocdbgreset_i ;
  + PROPERTY ACCESS_DIRECTION 0 top STRING ;

...

[ERROR  ] node 632857 (8323.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site
[ERROR  ] node 632858 (8523.29, 3746.29) failed to align to row 311 (3746.29, 3758.29) and site
[ERROR  ] node 632861 (8718.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site
[ERROR  ] node 632872 (8267.29, 3746.29) failed to align to row 311 (3746.29, 3758.29) and site
[ERROR  ] node 632877 (8402.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site
[ERROR  ] node 632911 (8289.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site
[INFO   ] Legality check takes 8303.42 ms
[INFO   ] DREAMPlace - ABCDPlace input legal flag = 0
[INFO   ] DREAMPlace - detailed placement takes 8.307 seconds
[INFO   ] DREAMPlace - iteration  712, HPWL 1.285793E+08, time 0.475ms
[INFO   ] DREAMPlace - non-linear placement takes 45.47 seconds
[INFO   ] DREAMPlace - writing to results/design/design.gp.def
[INFO   ] writing placement to results/design/design.gp.def
[INFO   ] DREAMPlace - write SolutionFileFormat.DEF takes 4.106 seconds
[INFO   ] DREAMPlace - placement takes 70.298 seconds
```

Would you give me some advice to handle this problem?
Thanks.

WOOSHIK-M, 2022-03-30:
What are the `move node` and `failed to align ~~`?
I think they consume too much time to run DREAMPlace considering the size of the netlist.

limbo018, 2022-03-30:
Hi, the LEF/DEF parsers are naive versions that cannot support full LEF/DEF features, so you may need to tweak the files a bit to remove unsupported syntax. 

For the errors, when you encounter many errors, it will keep printing all of them. So the placer was just printing, but not running any algorithm. 

According to the log, it looks like the placer failed at the legalization step. It cannot find legal locations for all cells (nodes). Can you check how large is the overflow at the end of the global placement iterations?

WOOSHIK-M, 2022-03-30:
I appreciate your reply.
It is the full log of the DREAMPlace when using LEF/DEF files of our chip design.

[CORTEX_A5_DREAMPlace_log.txt.zip](https://github.com/limbo018/DREAMPlace/files/8378333/CORTEX_A5_DREAMPlace_log.txt.zip)

I don't understand why cannot find legal locations... Please check the log.

Also, is their any difference between Boolshelf format and LEF/DEF format in terms of run-time?

limbo018, 2022-03-31:
Hi, I have roughly located the problem. It failed at legalization (after global placement). It looks like your design contains many multi-row cells, which caused the problem in legalization. Could you check how many multi-row cells in your design? And what is their maximum height in term of rows?

WOOSHIK-M, 2022-04-01:
Thanks for your reply.

I requested the detailed design specs to our cooperations 
If there is multi-row cells in our design, how can handle this issue?

limbo018, 2022-04-02:
Hi, the algorithm supports multi-row cells, so I think it might be some bug there. Ideally, it is better to have the testcase for debugging, but I may first try to look for problems with side information.

WOOSHIK-M, 2022-04-04:
I will comment about the design information when the cooperators give me that.

I checked the latest commit you updated, but it doesn't work and it still occurs errors with the below message.
```[ERROR  ] node 632679 (8558.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site```  

I really hope to use your DREAMPlace. 
Thanks!

WOOSHIK-M, 2022-04-04:
I checked our design doesn't contain any `multi-row cells`...
Are there any other reasons that cause this error?

limbo018, 2022-04-04:
That's good to know. I guess some numerical issues cause the placer regard single-row cells as multi-row cells during legalization. I will try looking into it this week.

limbo018, 2022-04-05:
Hi, I have mostly located the problem, but I need a little bit more information to double-check it. Could you show me the ```DIEAREA``` in the original DEF file? This sentence usually locates at the beginning of the DEF file.  I found the site width is 140, but the row width does not contain integral numbers of sites, which can cause potential numerical issues in legalization.

WOOSHIK-M, 2022-04-06:
I found that `DIEAREA` in DEF file is below.

`DIEAREA ( 0 0 ) ( 0 1800000 ) ( 3250000 1800000 ) ( 3250000 0 ) ;`

Can I use your placer normally if it is fixed?
Thanks!

limbo018, 2022-04-06:
This is not a typical syntax when we describe DIEAREA. Can you change it to following and check what the log looks like?
```
DIEAREA ( 0 0 ) ( 3250000 1800000 ) ; 
```

WOOSHIK-M, 2022-04-06:
I modified the DEF file as your advice, but it still outputs a lot of `[ERROR  ] node 632666 (8756.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site` error...

limbo018, 2022-04-06:
That's expected, because the site width has an annoying number of 140, which can cause numerical issues when normalized to site width = 1. I'm wondering do you still see many message like following?

```
[DEBUG  ] 2. move node 170249 from 16396.3 to 16375.3, blank (4444.29, 16396.3)\
[DEBUG  ] 2. move node 163864 from 16392.3 to 16371.3, blank (4444.29, 16392.3)\
[DEBUG  ] 1. move node 549332 from 16382.3 to 16391.3, blank (16391.3, 16415.3)\
...
```

WOOSHIK-M, 2022-04-06:
I still have a large amount of the below error.

```
[ERROR  ] node 299085 (10811.3, 9890.29) failed to align to row 823 (9890.29, 9902.29) and site
[ERROR  ] node 299086 (10777.3, 9926.29) failed to align to row 826 (9926.29, 9938.29) and site
[ERROR  ] node 299087 (10767.3, 9866.29) failed to align to row 821 (9866.29, 9878.29) and site
...
```

WOOSHIK-M, 2022-04-06:
If there is any other necessary information about our design, please contact me!!!
Thanks!

limbo018, 2022-04-06:
Thanks. Do you mind uploading the new log (or send me through email)? I can probably get more information from the log.

limbo018, 2022-04-06:
It would also be good if you can provide the sentence of the lowest ROW and highest ROW. Basically, I would like to check the (x, y) coordinates of rows.

WOOSHIK-M, 2022-04-07:
```
die area = (14.2857, 14.2857, 23199.3, 12842.3) 2.97417e+08
row height = 12, site width = 1
num_bins = 2048x1024, bin sizes = 0.9434x1.04395
#pins = 2328700, #movable_pins = 2305180
total_movable_node_area = 8.21663e+07, total_fixed_node_area = 4.44939e+08, total_space_area = 1.5113e+08
utilization = 0.54368, target_density = 1
use placeable_area = 1.5113e+08 to compute fillers
total_filler_node_area = 6.89636e+07, #fillers = 563320, filler sizes = 10.202x12
```

I see that the `die area` is different from the one in the `DEF` file.
How can I get the (x, y) coordinates of rows? Is it also in the `DEF` file?

limbo018, 2022-04-07:
Hi, the ROW definition looks like following. It usually locates after the definition of ```DIEAREA```. 

[image]

Can you grab the lines for the ```top ROW``` and the ```bottom ROW```?

WOOSHIK-M, 2022-04-07:
Here!!
```
DIEAREA ( 0 0 ) ( 3250000 1800000 ) ;
ROW unit_row_0 unit 2000 2000 N DO 23185 BY 1 STEP 140 0 ;
ROW unit_row_1 unit 2000 3680 FS DO 23185 BY 1 STEP 140 0 ;
ROW unit_row_2 unit 2000 5360 N DO 23185 BY 1 STEP 140 0 ;
ROW unit_row_3 unit 2000 7040 FS DO 23185 BY 1 STEP 140 0 ;
ROW unit_row_4 unit 2000 8720 N DO 23185 BY 1 STEP 140 0 ;
ROW unit_row_5 unit 2000 10400 FS DO 23185 BY 1 STEP 140 0 ;
...
```

limbo018, 2022-04-07:
Hi, can you try this commit (https://github.com/limbo018/DREAMPlace/commit/1955989851621e637b3115ea2138546606d0ee91)? I shifted the coordinate system to make the lower-left origin of the layout (0, 0). It is supposed to fix the problem. Let me know whether it works or not. I will merge the commit to the master branch later.

WOOSHIK-M, 2022-04-07:
I still got the same error on that commit...

```
[ERROR  ] node 632871 (8384.29, 3758.29) failed to align to row 312 (3758.29, 3770.29) and site
[ERROR  ] node 632872 (8269.29, 3746.29) failed to align to row 311 (3746.29, 3758.29) and site
[ERROR  ] node 632874 (8500.29, 3746.29) failed to align to row 311 (3746.29, 3758.29) and site
[ERROR  ] node 632877 (8401.29, 3746.29) failed to align to row 311 (3746.29, 3758.29) and site
[ERROR  ] node 632911 (8312.29, 3746.29) failed to align to row 311 (3746.29, 3758.29) and site
...
```

limbo018, 2022-04-07:
Did you ```make install``` again before running?

WOOSHIK-M, 2022-04-07:
OMG, I forgot that. I tell you it works or not after re-`make install`.

WOOSHIK-M, 2022-04-07:
I finally got the normal message!! I really appreciate your help.