limbo018/DREAMPlace #70: LEF/DEF unit

hathubkhn, 2022-03-31:
Hello Limbo, 
Thank you for this repo. It is really helpful in designing chips. However, I see some problems when LEF unit = 2000 while DEF unit = 1000 ( 2 files are downloaded from the repo: https://github.com/ahmed-agiza/EDAViewer/blob/master/server/example/Nangate45/). The written def output may not be correct. And some cell is out of boundary. 

I add my folder which includes:
1. Log 
2. lef-def file
3. outline of output and original. ( I used edaviewer.com website)

Could you help me to figure out what is the problem? Thank you very much[
[file.zip](https://github.com/limbo018/DREAMPlace/files/8389487/file.zip)
](url)

limbo018, 2022-03-31:
Hi, can you set the num_bins in JSON to 64x64 or 128x128? As your benchmark is very small, I think the problem comes from the last step of legalization.

hathubkhn, 2022-03-31:
Thank for your rely. I try to change num_bins to 64x64. However, I have another problem
[DEBUG  ] Standard cell legalization num_unplaced_cells = 1
[INFO   ] Greedy legalization takes 0.122141 ms
[ERROR  ] DREAMPlace - legality check failed in greedy legalization
[INFO   ] DREAMPlace - legalization takes 0.001 seconds
[DEBUG  ] node 161 (-242.199, 62.5722, -226.199, 69.9406) out of boundary
[ERROR  ] node 161 (-242.199, 62.5722) failed to align to row 8 (58.9474, 66.3158), gap 3.62484
[ERROR  ] node 161 (-242.199, 62.5722) failed to align to row 8 (58.9474, 66.3158) and site 
[INFO   ] Legality check takes 0.026021 ms
[INFO   ] DREAMPlace - iteration  288, HPWL 2.033594E+04, time 0.106ms
[INFO   ] DREAMPlace - Start ABCDPlace for refinement
[INFO   ] DREAMPlace - ABCDPlace input legal flag = 0
[INFO   ] DREAMPlace - detailed placement takes 0.000 seconds
[DEBUG  ] node 161 (-242.199, 62.5722, -226.199, 69.9406) out of boundary
[ERROR  ] node 161 (-242.199, 62.5722) failed to align to row 8 (58.9474, 66.3158), gap 3.62484
[ERROR  ] node 161 (-242.199, 62.5722) failed to align to row 8 (58.9474, 66.3158) and site 
[INFO   ] Legality check takes 0.02216 ms
[INFO   ] DREAMPlace - iteration  289, HPWL 2.033594E+04, time 0.099ms
[INFO   ] DREAMPlace - non-linear placement takes 2.63 seconds
[INFO   ] DREAMPlace - writing to results/gcd_unplaced/gcd_unplaced.gp.def
thu 380.0
[INFO   ] DREAMPlace - write SolutionFileFormat.DEF takes 0.003 seconds
[WARNING] DREAMPlace - External detailed placement engine thirdparty/ntuplace3 or aux file NOT found
[INFO   ] DREAMPlace - placement takes 2.658 seconds
[INFO   ] writing placement to results/gcd_unplaced/gcd_unplaced.gp.def

hathubkhn, 2022-03-31:
And I have another question. Because of lef_unit/def_unit = 2, STEP changed from 190 to 380. However, die area and the number of steps in each row are the same as DEF input. I thought it should be changed from 120 to 60, or die area should be the double size. Is it true?
===> DEF input
ROW CORE_ROW_0 FreePDK45_38x28_10R_NP_162NW_34O 0 0 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_1 FreePDK45_38x28_10R_NP_162NW_34O 0 1400 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_2 FreePDK45_38x28_10R_NP_162NW_34O 0 2800 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_3 FreePDK45_38x28_10R_NP_162NW_34O 0 4200 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_4 FreePDK45_38x28_10R_NP_162NW_34O 0 5600 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_5 FreePDK45_38x28_10R_NP_162NW_34O 0 7000 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_6 FreePDK45_38x28_10R_NP_162NW_34O 0 8400 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_7 FreePDK45_38x28_10R_NP_162NW_34O 0 9800 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_8 FreePDK45_38x28_10R_NP_162NW_34O 0 11200 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_9 FreePDK45_38x28_10R_NP_162NW_34O 0 12600 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_10 FreePDK45_38x28_10R_NP_162NW_34O 0 14000 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_11 FreePDK45_38x28_10R_NP_162NW_34O 0 15400 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_12 FreePDK45_38x28_10R_NP_162NW_34O 0 16800 FS DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_13 FreePDK45_38x28_10R_NP_162NW_34O 0 18200 N DO 120 BY 1 STEP 190 0
 ;
ROW CORE_ROW_14 FreePDK45_38x28_10R_NP_162NW_34O 0 19600 FS DO 120 BY 1 STEP 190 0

====> DEF output
DIEAREA ( 0 0 ) ( 22800 21000 ) ;

ROW CORE_ROW_0 FreePDK45_38x28_10R_NP_162NW_34O 0 0 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_1 FreePDK45_38x28_10R_NP_162NW_34O 0 2800 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_2 FreePDK45_38x28_10R_NP_162NW_34O 0 5600 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_3 FreePDK45_38x28_10R_NP_162NW_34O 0 8400 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_4 FreePDK45_38x28_10R_NP_162NW_34O 0 11200 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_5 FreePDK45_38x28_10R_NP_162NW_34O 0 14000 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_6 FreePDK45_38x28_10R_NP_162NW_34O 0 16800 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_7 FreePDK45_38x28_10R_NP_162NW_34O 0 19600 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_8 FreePDK45_38x28_10R_NP_162NW_34O 0 22400 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_9 FreePDK45_38x28_10R_NP_162NW_34O 0 25200 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_10 FreePDK45_38x28_10R_NP_162NW_34O 0 28000 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_11 FreePDK45_38x28_10R_NP_162NW_34O 0 30800 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_12 FreePDK45_38x28_10R_NP_162NW_34O 0 33600 FS DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_13 FreePDK45_38x28_10R_NP_162NW_34O 0 36400 N DO 120 BY 1 STEP 380 0 
;
ROW CORE_ROW_14 FreePDK45_38x28_10R_NP_162NW_34O 0 39200 FS DO 120 BY 1 STEP 380 0

hathubkhn, 2022-03-31:
Thank you very much ^^

limbo018, 2022-04-01:
Hi, bug fixed in this commit https://github.com/limbo018/DREAMPlace/commit/b88b7b250e828ff8e66992eff2fb4bfa58c6f59c

hathubkhn, 2022-04-01:
Yes, I build code again and the problem is fixed. Thank you very much to take your time.

hathubkhn, 2022-04-07:
Hello Prof.Lim, 
As I understand in your program, you want to minimize overflow and get best_pos. After that, you will do next step that is re-order. So, could I ask you 2 questions:
1. In my case, I want to minimize congestion and wirelength. How can I change the objective function? 
2. Which part in your program will do re-order? I try to figure out but I dont know exactly where is it in? 
Thank you so much!

limbo018, 2022-04-07:
Hi, in the README, there is a figure indicating the flow of the algorithm. It is more complicated than your description, but you get the first step correct. 

For steps in detailed placement, each is an operator located in the ```dreamplace/ops/``` folder.