limbo018/DREAMPlace #72: TypeError: 'NoneType' object is not callable

Eternity666, 2022-04-10:
Hi, sorry to bother! 
At the end of the iteration process, I want to call ```model.op_collections.pin_utilization_map_op(pos)``` and ```model.op_collections.route_utilization_map_op(pos)``` to collect data, but these two functions will raise **TypeError** when I use data in ISPD2005. 
It's worth noting that the ```Placer.py``` runs successfully with all superblue in DAC2012, but it raises the error when running with data in ISPD2005.
Do you know the solution to this problem? Thank you very much!

My command is 
```python ./dreamplace/Placer.py ./test/ispd2005/adaptec1.json```
And the screenshot detailing the error is shown here.
[image]

limbo018, 2022-04-11:
This is expected. If you compare the files in the two benchmarks, ISPD 2005 does not contain routing information, so you cannot run ISPD 2005 for routability optimization.

Eternity666, 2022-04-18:
Got it! Thank you very much! And May I ask how to obtain the HPWL of each net in a netlist?

limbo018, 2022-04-18:
There is a CPU implementation here for your reference. 
https://github.com/limbo018/DREAMPlace/blob/6cf549fc6e4905267f336f9689ab72ccb775a6e0/dreamplace/PlaceDB.py#L341

It is rather slow and does not work with PyTorch tensors. You may take it as a reference and implement your own version. 

The HPWL operator in the dreamplace/ops folder computes total HPWL, which is accelerated on GPU, but it cannot give you the HPWL of a specific net.

Eternity666, 2022-04-18:
Thanks a lot! But I still have a question, that is, why does it has to multiply net_weights after computing the half perimeter?
https://github.com/limbo018/DREAMPlace/blob/6cf549fc6e4905267f336f9689ab72ccb775a6e0/dreamplace/PlaceDB.py#L353
Above is different from the code in ```hpwl.cpp```
https://github.com/limbo018/DREAMPlace/blob/6cf549fc6e4905267f336f9689ab72ccb775a6e0/dreamplace/ops/hpwl/src/hpwl.cpp#L77

limbo018, 2022-04-18:
Net weights are optional. You can set it to all ones.