limbo018/DREAMPlace #73: COULD NOT FIND CUDA

WOOSHIK-M, 2022-04-14:
Hi!
I also got an error when installing DREAMPlace after adding GPUs.

The below figure shows the version of drivers and CUDA.
[image]

And, The below error message occurs when I implement 'make'.
[image]

I checked if CUDA is available w/ torch, but it is run normally.
```python
$  python -c "import torch; print(torch.cuda.is_available())"
$ True
"""

How can I handle this issue?

gzz2000, 2022-04-14:
Hi, I'm a student of Prof. Lin. This error is due to a newer system CUB version not correctly handled. We have fixed the bug in the newest commit. Please try pulling the new commit, and re-`make install` and see if the problem persists.

WOOSHIK-M, 2022-04-14:
I appreciate your reply. However, another error occurs after pulling a new commit and `make install`.
Any error did not occur but the log is unnormal and the code is terminated unnormally. 
The error message is described below.

```
...

[INFO   ] DREAMPlace - iteration   95, (  95,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.279ms
[INFO   ] DREAMPlace - full step 15.527 ms
[INFO   ] DREAMPlace - optimizer step 14.739 ms
[INFO   ] DREAMPlace - iteration   96, (  96,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.292ms
[INFO   ] DREAMPlace - full step 15.451 ms
[INFO   ] DREAMPlace - optimizer step 15.062 ms
[INFO   ] DREAMPlace - iteration   97, (  97,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.279ms
[INFO   ] DREAMPlace - full step 15.749 ms
[INFO   ] DREAMPlace - optimizer step 14.618 ms
[INFO   ] DREAMPlace - iteration   98, (  98,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.279ms
[INFO   ] DREAMPlace - full step 15.328 ms
[INFO   ] DREAMPlace - optimizer step 14.640 ms
[INFO   ] DREAMPlace - iteration   99, (  99,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.309ms
[INFO   ] DREAMPlace - full step 15.364 ms
[INFO   ] DREAMPlace - optimizer step 14.550 ms
[INFO   ] DREAMPlace - iteration  100, ( 100,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.278ms
[INFO   ] DREAMPlace - full step 15.255 ms
[INFO   ] DREAMPlace - optimizer step 14.452 ms
[INFO   ] DREAMPlace - iteration  101, ( 101,  0,  0), Obj NAN, DensityWeight NAN, HPWL 0.000000E+00, Overflow 0.000000E+00, MaxDensity 0.000E+00, gamma 1.000951E+01, time 0.278ms
[INFO   ] DREAMPlace - full step 15.137 ms
[INFO   ] DREAMPlace - optimizer nesterov takes 2.104 seconds
[INFO   ] DREAMPlace - Start legalization
[INFO   ] Macro legalization: regard 0 cells as dummy fixed (movable macros)
[INFO   ] Macro legalization takes 0.223727 ms
[DEBUG  ] Standard cell legalization blank_num_bins_y = 890
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 890
[INFO   ] Standard cell legalization legalizeBin takes 2307.408 ms
[DEBUG  ] Standard cell legalization num_unplaced_cells = 210904
[DEBUG  ] Standard cell legalization blank_num_bins_y = 890
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 890
[INFO   ] Standard cell legalization legalizeBin takes 2290.416 ms
[DEBUG  ] Standard cell legalization num_unplaced_cells = 210904
[INFO   ] Greedy legalization takes 4547.33 ms
[INFO   ] Legality check takes 2.65079 ms
[ASSERT ] /home/wooshik.myung/MRX/test_code/DREAMPlace/dreamplace/ops/abacus_legalize/../abacus_legalize/src/abacus_legalize_cpu.h:147: DreamPlace::abacusPlaceRowCPU(const T*, const T*, const T*, const T*, T*, T, T, T, int, int, int*, DreamPlace::AbacusCluster<T>*, int) [with T = float]::<lambda(int, float, float)>: Assertion `cluster->x >= range_xl && cluster->x + cluster->w <= range_xh' failed
Aborted (core dumped)
```

gzz2000, 2022-04-14:
Can you paste your cmake log? We suspect this might be a problem related to cuda compilation flags.

WOOSHIK-M, 2022-04-14:
I downgraded CUDA to 11.4 and the cmake message is as below.

[image]

```
-- The CXX compiler identification is GNU 8.4.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- CMAKE_INSTALL_PREFIX: /home/wooshik.myung/MRX/test_code/DREAMPlace/install
-- CMAKE_BUILD_TYPE: Release
-- CMAKE_CXX_ABI: _GLIBCXX_USE_CXX11_ABI=0
-- pybind11 v2.6.3 dev1
-- Found PythonInterp: /home/wooshik.myung/anaconda3/envs/test_dp/bin/python (found version "3.8.13")
-- Found PythonLibs: /home/wooshik.myung/anaconda3/envs/test_dp/lib/libpython3.8.so
-- Performing Test HAS_FLTO
-- Performing Test HAS_FLTO - Success
-- TORCH_INSTALL_PREFIX=/home/wooshik.myung/anaconda3/envs/test_dp/lib/python3.8/site-packages/torch
-- TORCH_VERSION=1.9
-- Found CUDA: /usr/local/cuda (found suitable version "11.4", minimum required is "9.0")
-- TORCH_ENABLE_CUDA=1
-- Found OpenMP_CXX: -fopenmp (found version "4.5")
-- Found OpenMP: TRUE (found version "4.5")
-- Found ZLIB: /usr/lib/x86_64-linux-gnu/libz.so (found version "1.2.11")
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.55.0")
-- Boost_INCLUDE_DIRS = /usr/include
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'cairo'
--   Found cairo, version 1.15.10
-- Found Cairo: /usr/include/cairo
-- Cairo: /usr/include/cairo
-- Cairo: /usr/lib/x86_64-linux-gnu/libcairo.so
-- OPS_DIR /home/wooshik.myung/MRX/test_code/DREAMPlace/dreamplace/ops
-- UTILITY_LIBRARY_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/install/dreamplace/ops/utility
-- FLUTE_INCLUDE_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/thirdparty/flute-3.1
-- FLUTE_LINK_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/install/thirdparty/flute-3.1
-- LIMBO_SOURCE_DIR /home/wooshik.myung/MRX/test_code/DREAMPlace/thirdparty/Limbo
-- LIMBO_BINARY_DIR /home/wooshik.myung/MRX/test_code/DREAMPlace/install/thirdparty/Limbo
-- CUB_DIR /usr/local/cuda/include
-- MUNKRES_CPP_INCLUDE_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/thirdparty/munkres-cpp/src
-- MUNKRES_CPP_LINK_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/install/thirdparty/munkres-cpp
-- LEMON_INCLUDE_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/thirdparty/Limbo/limbo/thirdparty/lemon;/home/wooshik.myung/MRX/test_code/DREAMPlace/install/thirdparty/Limbo/limbo/thirdparty/lemon
-- LEMON_LINK_DIRS /home/wooshik.myung/MRX/test_code/DREAMPlace/install/thirdparty/Limbo/limbo/thirdparty/lemon/lemon
-- CUDA_ARCH_FLAGS: -gencode;arch=compute_60,code=sm_60;-gencode;arch=compute_61,code=sm_61;-gencode;arch=compute_70,code=sm_70;-gencode;arch=compute_75,code=sm_75;-gencode;arch=compute_80,code=sm_80;-gencode;arch=compute_86,code=sm_86
-- The C compiler identification is GNU 8.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- CMAKE_BUILD_TYPE: Release
-- CMAKE_CXX_ABI: _GLIBCXX_USE_CXX11_ABI=0
-- OPENBLAS: OFF
-- INSTALL_LIMBO: OFF
-- GENERATE_DOCS: OFF
-- ENABLE_TEST: OFF
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Found BISON: /usr/local/bin/bison (found suitable version "3.4", minimum required is "3.3")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Found Boost: /usr/include (found suitable version "1.65.1", minimum required is "1.55.0") found components: graph regex
-- Performing Test LPSOLVE_LINKS_ALONE
-- Performing Test LPSOLVE_LINKS_ALONE - Failed
-- Performing Test LPSOLVE_LINKS_WITH_EXTRA_LIBS
-- Performing Test LPSOLVE_LINKS_WITH_EXTRA_LIBS - Success
-- Found LpSolve: /usr/lib/liblpsolve55.a;/usr/lib/x86_64-linux-gnu/libdl.so;/usr/lib/x86_64-linux-gnu/libcolamd.so
CMake Deprecation Warning at thirdparty/Limbo/limbo/thirdparty/lemon/CMakeLists.txt:4 (CMAKE_POLICY):
  The OLD behavior for policy CMP0048 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.

-- Found Wget: /usr/bin/wget
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.13") found components: doxygen dot
-- Found Ghostscript: /usr/bin/gs
-- Could NOT find COIN (missing: COIN_INCLUDE_DIR COIN_CBC_LIBRARY COIN_CBC_SOLVER_LIBRARY COIN_CGL_LIBRARY COIN_CLP_LIBRARY COIN_COIN_UTILS_LIBRARY COIN_OSI_LIBRARY COIN_OSI_CBC_LIBRARY COIN_OSI_CLP_LIBRARY)
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of long long
-- Check size of long long - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/wooshik.myung/MRX/test_code/DREAMPlace/install
```

I go to try to run on this environment.

WOOSHIK-M, 2022-04-14:
I re-installed CUDA and nvidia driver and it works normally.
Thanks for your help!

closed 2022-04-14 by WOOSHIK-M