limbo018/DREAMPlace #74: Location information

WOOSHIK-M, 2022-04-17:
Hi professor Lin.
I'm using your DREAMPlace for my own project, and want to know how to operate the `PlaceDB` class.
Because `ICC2` which is EDA-Tool I'm using only provides `LEF/DEF` format, not `BOOKSHELF`, thus, I'm trying to know the process of converting from `LEF/DEF` to `PlaceDB`.

### The point is ...
How can I get the location information of the `FIXED` macros in `PlaceDB` class? I checked the `node_x` and `node_y` attributes using `node_name2id_map`, but the values are different from those in `LEF/DEF` files.

WOOSHIK-M, 2022-04-17:
I also checked `node_size_x` and `node_size_y` are totally different... How can I get the information in `PlaceDB` of the particular macro? Does not `node` in this class indicate `macro` or `std cell` in the `LEF/DEF`?

limbo018, 2022-04-17:
The coordinate system is shifted and scaled. Please search ```coordinate system``` to find the shift and scale factor. You need to convert back to match the coordinate system in the LEF/DEF.

WOOSHIK-M, 2022-04-17:
Thank you for the reply.
Could you explain what is the `coordinate system`? 
I see that the scaling factors between `placedb.node_size_x` and the width of each macro...

Which file contains this process? (scaling and shifting)

WOOSHIK-M, 2022-04-18:
Hi, I googled the `coordinate system` yesterday but didn't find any information about that.
Does not the `coordinate system` be defined by you?

Thank you.

limbo018, 2022-04-18:
Hi, it is not something special. You may check the ```scale``` function in ```PlaceDB.py``` (https://github.com/limbo018/DREAMPlace/blob/6cf549fc6e4905267f336f9689ab72ccb775a6e0/dreamplace/PlaceDB.py#L145). The shift and scale factors are printed in the log, so you can scale back to the original coordinates/sizes to match the values defined in ```LEF/DEF```

WOOSHIK-M, 2022-04-18:
OMG!
I really appreciate your help!!!

Why do we need `scale_factor` and `shift_factor`? What do they do?

I see that the coordinates in `LEF/DEF` files provided by ICC2 are the `bottom_left` of the macros. Is it right? Don't we need to convert from `bottom_left` to `center`?

limbo018, 2022-04-18:
Yes. The coordinates are lower-left corner. The algorithm assumes site width is 1, so we need to scale the coordinates.

WOOSHIK-M, 2022-04-18:
I have the last question!! 

Is it enough to only modify `placedb.node_x` and `placedb.node_y` to evaluate the different macro placements on the same design?

limbo018, 2022-04-18:
placedb.node_x and placedb.node_y only control the initial positions. If you want to have more detailed control, you can also look into data_collections.pos which is used all over the optimization.

WOOSHIK-M, 2022-04-19:
Hi professor!
Could I turn off the logging message? if it can, how?