limbo018/DREAMPlace #75: RUDY implementation

HAOHANZHOU, 2022-04-20:
Hi, does dreamplace/ops/rudy implements RUDY metric mentioned in paper: 
Fast and Accurate Routing Demand Estimation for Efficient Routability-driven Placement
https://ieeexplore.ieee.org/document/4211973
It seems there is no horizontal or vertical congestion concept mentioned in article, but I found them in implementation.
[image]

limbo018, 2022-08-13:
Hi, the RUDY implementation is more based on the following paper.
Cheng, Chih-Liang Eric. "RISA: Accurate and efficient placement routability modeling." Proceedings of the 1994 IEEE/ACM international conference on Computer-aided design. 1994.

You are very welcome to contribute new OPs and extend the routability estimation.

YihangQiu, 2022-11-08:
Hi! Prof. Lin. 

I don't know how to eliminate the randomness caused by the _atomicAdd()_ function in _rudy_cuda_kernel.cu._ 
I would like the program to run twice with the same result when I use the route utilization map generated by rudy.

Thanks for your reply!
[image]

limbo018, 2022-11-09:
Hi, there are commits to make rudy deterministic in another branch, which is an ongoing task. I can pick the commits to master branch in today or tomorrow.

limbo018, 2022-11-09:
Hi, please checkout develop branch for this commit (https://github.com/limbo018/DREAMPlace/commit/5199a626314e7e390c593ffa248efe633b37827f). We have added a ```deterministic_flag``` to RUDY as well. Remember to turn ```deterministic_flag``` to 1 in json.

YihangQiu, 2022-11-09:
Okay, thank you Prof. Lin.

ybiao-he, 2023-01-19:
Hi Dr. Lin,

I just went through the paper (Cheng, Chih-Liang Eric. "RISA: Accurate and efficient placement routability modeling." Proceedings of the 1994 IEEE/ACM international conference on Computer-aided design. 1994.). It seems that they consider the existing wire and blockage while computing the congestion value, they should reduce the routing resource supply, but in the implementation, I did not find any information of existing wires and blockage in the input of RUDY function/class. Are the existing wires and blockages considered in this implementation?

limbo018, 2023-01-19:
Fixed macros are handled by ```initial_horizontal_utilization_map``` and ```initial_vertical_utilization_map``` in our implementation. If you trace these variables, they are initialized in ```place_io```.

ybiao-he, 2023-01-19:
i.c. Thanks very much!

closed 2024-06-06 by HAOHANZHOU