limbo018/DREAMPlace #81: DEF file legalization failed

dearmrlv, 2022-07-31:
Hi, Professor Lin. This is an awesome project. And I encountered some problem when using it. 
I tried to use DREAMPlace to perform global placement for my project, but error appeared, and the log is showed below:
```
[INFO   ] lef_input = test/tcbn45gsbwp_9lm7X1ZRDL.lef
[INFO   ] def_input = test_def/fp.def
[INFO   ] verilog_input = ICC_results/APCA_0.25r_0.45u/top-compiled.v
[INFO   ] bookshelf_aux_input =
[INFO   ] bookshelf_pl_input =
[INFO   ] def_size_input =
[INFO   ] def_output = fp-out.def
[INFO   ] rpt_output =
[INFO   ] target_util = 0
[INFO   ] max_displace = 0
[INFO   ] bin size = (10, 10) #rows
[INFO   ] sbin size = (5, 5) #rows
[INFO   ] bin_space_threshold = 0.2
[INFO   ] abu = 2,10 5,5 10,2 20,1
[INFO   ] def_ignore_cells =
[INFO   ] macro_obs_aware_layers =
[INFO   ] enable_place = true
[INFO   ] enable_legalize = true
[INFO   ] evaluate_overlap = false
[INFO   ] move_multi_row_cell = true
[INFO   ] align_power_line = true
[INFO   ] cluster_cell = false
[INFO   ] sort_nets_by_degree = false
[INFO   ] file_format = DEF
[INFO   ] max_iters = 6
[INFO   ] reading test/tcbn45gsbwp_9lm7X1ZRDL.lef
[INFO   ] set CORE site to core, 280 x 2520
[INFO   ] set CORE site to bcore, 280 x 5040
[INFO   ] set CORE site to bcoreExt, 280 x 5040
[INFO   ] set CORE site to ccore, 280 x 7560
[INFO   ] set CORE site to dcore, 280 x 10080
[INFO   ] set CORE site to gacore, 1400 x 2520
[INFO   ] set CORE site to gabcore, 1400 x 5040
[INFO   ] reading test_def/fp.def
[INFO   ] detect 149 rows, 51809 components, 4354 IO pins, 56099 nets, 0 blockages
TECHNOLOGY TSMC_N40_9lm7X1ZRDL_TCBN40 ;
Traceback (most recent call last):
  File "run_DREAMPlace.py", line 5, in <module>
    db, params = read_db(sys.argv[1])
  File "/lustre/S/lvzhengyang/CA/read_db.py", line 37, in read_db
    placedb(params)
  File "/workspace/S/lvzhengyang/DREAMPlace_Py38_2/dreamplace/PlaceDB.py", line 552, in __call__
    self.read(params)
  File "/workspace/S/lvzhengyang/DREAMPlace_Py38_2/dreamplace/PlaceDB.py", line 435, in read
    self.rawdb = place_io.PlaceIOFunction.read(params)
  File "/workspace/S/lvzhengyang/DREAMPlace_Py38_2/dreamplace/ops/place_io/place_io.py", line 35, in read
    return place_io_cpp.forward(args.split(' '))
IndexError: _Map_base::at
```
And here is my config, LEF and DEF file:
The DEF file is generated by ICC 2018.
[lefdef.zip](https://github.com/limbo018/DREAMPlace/files/9228586/lefdef.zip)

Thanks so much to have a look at my issue, and I'll be greatly appreciated for your suggestions!

limbo018, 2022-08-01:
Hi, I found one problem is that the ROW definition in DEF. The row macro name "unit" is inconsistent with the site names in LEF. You need to replace it to "core".

dearmrlv, 2022-08-01:
Thank you Professor! Your suggestion worked, and solved the above-mentioned error. However, I met another error and the log is below. Could you provide some suggestions on it?
```
[INFO   ] lef_input = test/tcbn45gsbwp_9lm7X1ZRDL.lef
[INFO   ] def_input = test_def/fp.def
[INFO   ] verilog_input = ICC_results/APCA_1.0r_0.55u/top-compiled.v
[INFO   ] bookshelf_aux_input =
[INFO   ] bookshelf_pl_input =
[INFO   ] def_size_input =
[INFO   ] def_output = fp-out.def
[INFO   ] rpt_output =
[INFO   ] target_util = 0
[INFO   ] max_displace = 0
[INFO   ] bin size = (10, 10) #rows
[INFO   ] sbin size = (5, 5) #rows
[INFO   ] bin_space_threshold = 0.2
[INFO   ] abu = 2,10 5,5 10,2 20,1
[INFO   ] def_ignore_cells =
[INFO   ] macro_obs_aware_layers =
[INFO   ] enable_place = true
[INFO   ] enable_legalize = true
[INFO   ] evaluate_overlap = false
[INFO   ] move_multi_row_cell = true
[INFO   ] align_power_line = true
[INFO   ] cluster_cell = false
[INFO   ] sort_nets_by_degree = false
[INFO   ] file_format = DEF
[INFO   ] max_iters = 6
[INFO   ] reading test/tcbn45gsbwp_9lm7X1ZRDL.lef
[INFO   ] set CORE site to core, 280 x 2520
[INFO   ] set CORE site to bcore, 280 x 5040
[INFO   ] set CORE site to bcoreExt, 280 x 5040
[INFO   ] set CORE site to ccore, 280 x 7560
[INFO   ] set CORE site to dcore, 280 x 10080
[INFO   ] set CORE site to gacore, 1400 x 2520
[INFO   ] set CORE site to gabcore, 1400 x 5040
[INFO   ] reading test_def/fp.def
[INFO   ] detect 269 rows, 51809 components, 4354 IO pins, 56099 nets, 0 blockages
TECHNOLOGY TSMC_N40_9lm7X1ZRDL_TCBN40 ;
[WARNING] no Bookshelf file specified
[WARNING] no additional Bookshelf .pl file specified
[WARNING] 0 nets with 0 pins from same nodes
[WARNING] 0 nets should be ignored due to not enough pins
[INFO   ] sort nodes in the order of movable and fixed
[INFO   ] Group cells for fence regions
[INFO   ] Construct 0 groups
[INFO   ] Fence region groups done
[DEBUG  ] num_terminals 0, numFixed 0, numPlaceBlockages 0, num_terminal_NIs 4354
[DEBUG  ] fixed area overlap: 0 fixed area total: 0, space area = 4.6066e+11
/workspace/S/lvzhengyang/DREAMPlace_Py38_2/dreamplace/PlaceDB.py:538: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  self.node2pin_map = np.array(self.node2pin_map)
/workspace/S/lvzhengyang/DREAMPlace_Py38_2/dreamplace/PlaceDB.py:543: VisibleDeprecationWarning: Creating an ndarray from ragged nested sequences (which is a list-or-tuple of lists-or-tuples-or ndarrays with different lengths or shapes) is deprecated. If you meant to do this, you must specify 'dtype=object' when creating the ndarray.
  self.net2pin_map = np.array(self.net2pin_map)
WARNING:root:net weights are all the same, ignored
WARNING:root:quadratic density penalty is disabled, density weight update is forced to be based on HPWL
[INFO   ] Macro legalization: regard 0 cells as dummy fixed (movable macros)
[INFO   ] Macro legalization takes 0.100158 ms
[DEBUG  ] Standard cell legalization blank_num_bins_y = 134
[DEBUG  ] Standard cell legalization iteration 0 with 1x1 bins
[DEBUG  ] Standard cell legalization #bin_blanks
[DEBUG  ] #bin_objs = 134
[DEBUG  ] 2. move node 9688 from 325.429 to 321.429, blank (71.4286, 325.429)
[DEBUG  ] 2. move node 6208 from 325.429 to 321.429, blank (71.4286, 325.429)
[DEBUG  ] 2. move node 17215 from 326.429 to 325.429, blank (71.4286, 326.429)
[DEBUG  ] 2. move node 47004 from 325.429 to 324.429, blank (71.4286, 325.429)
[DEBUG  ] 2. move node 47001 from 324.429 to 323.429, blank (71.4286, 324.429)
[DEBUG  ] 2. move node 5698 from 324.429 to 320.429, blank (71.4286, 324.429)
[DEBUG  ] 2. move node 44100 from 324.429 to 323.429, blank (71.4286, 324.429)
[DEBUG  ] 2. move node 17277 from 324.429 to 323.429, blank (71.4286, 324.429)
[DEBUG  ] 2. move node 31468 from 326.429 to 324.429, blank (71.4286, 326.429)
[DEBUG  ] 2. move node 44559 from 325.429 to 323.429, blank (71.4286, 325.429)
[... 66 lines cut ...]
[ERROR  ] row 133 (550.229, 553.829), overlap node 10446 (521.429, 550.229, 525.429, 552.029) with node 14865 (523.429, 550.229, 524.029, 552.029), gap 2
[ERROR  ] row 133 (550.229, 553.829), overlap node 17939 (525.429, 550.229, 526.029, 552.029) with node 47061 (525.429, 550.229, 526.229, 552.029), gap 0.599976
[ERROR  ] row 133 (550.229, 553.829), overlap node 9719 (526.429, 550.229, 530.429, 552.029) with node 46951 (526.429, 550.229, 527.229, 552.029), gap 4
[ERROR  ] row 133 (550.229, 553.829), overlap node 10216 (530.429, 550.229, 534.429, 552.029) with node 45981 (530.429, 550.229, 531.229, 552.029), gap 4
[ERROR  ] row 133 (550.229, 553.829), overlap node 14868 (534.429, 550.229, 535.029, 552.029) with node 47173 (534.429, 550.229, 535.229, 552.029), gap 0.599976
[ERROR  ] row 133 (550.229, 553.829), overlap node 14670 (535.429, 550.229, 536.029, 552.029) with node 47210 (535.429, 550.229, 536.229, 552.029), gap 0.599976
[ERROR  ] row 133 (550.229, 553.829), overlap node 14682 (536.429, 550.229, 537.029, 552.029) with node 47242 (536.429, 550.229, 537.229, 552.029), gap 0.599976
[ERROR  ] row 133 (550.229, 553.829), overlap node 17132 (537.429, 550.229, 538.029, 552.029) with node 43829 (537.429, 550.229, 538.029, 552.029), gap 0.599976
[... cut ...]
[ERROR  ] row 133 (550.229, 553.829), overlap node 8556 (551.429, 550.229, 555.429, 552.029) with node 44001 (551.429, 550.229, 552.229, 552.029), gap 4
[ERROR  ] row 133 (550.229, 553.829), overlap node 17130 (555.429, 550.229, 556.029, 552.029) with node 17909 (555.429, 550.229, 556.029, 552.029), gap 0.599976
[INFO   ] Legality check takes 1226.31 ms
```
Sorry to let you solve my puzzle again. I am appreciated for any of your suggestions!
Again, the lef and def is attached below.

[lefdef.zip](https://github.com/limbo018/DREAMPlace/files/9231560/lefdef.zip)

Best Wishes!

limbo018, 2022-08-01:
This problem has been fixed with the latest commit. Could you pull and reinstall?

dearmrlv, 2022-08-02:
Thank you so much! And my problem was finally solved. I'll close this issue .

closed 2022-08-02 by dearmrlv