limbo018/DREAMPlace #85: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs

SWang848, 2022-09-04:
Hey all,

I built and run the code in WSL. All dependencies and the whole cmake build process are finished without error.
I'm having troubles in running `python dreamplace/Placer.py test/ispd2005/adaptec1.json`

> ImportError: /mnt/c/Users/Owner/EDA-AI/DREAMPlace/dreamplace/ops/move_boundary/move_boundary_cpp.cpython-38-x86_64-linux-gnu.so: undefined symbol: _ZN3c106detail14torchCheckFailEPKcS2_jRKSs

There seems to be some error about an undefined symbol. Where is this defined? I saw the similar issues about the undefined symbol. I also tried to set -DCMAKE_CXX_ABI=1 as that issue suggested, but the problem is not solved. 

Do you have any ideas?

limbo018, 2022-09-04:
Hi, the code has not been tested under WSL, so there might be unexpected problems. According to the name of the symbol, it can be incompatibility of the PyTorch version. Can you try directly run python and import torch with simple tensor operations to see whether PyTorch works under your env?

SWang848, 2022-09-05:
> Hi, the code has not been tested under WSL, so there might be unexpected problems. According to the name of the symbol, it can be incompatibility of the PyTorch version. Can you try directly run python and import torch with simple tensor operations to see whether PyTorch works under your env?

Hi thanks for your quick response. I tried as you suggest that I run python and import torch with simple tensor opertaions. It run well and smoothly. I also tried the different version of pytorch-cpu, like 1.12.0(the latest), 1.8.0(suggested in readme), but it also has the similar undefined problem related with pytorch. Is it because torch GPU version is not installed, or anything else?

SWang848, 2022-09-06:
The problem solved!! When running `cmake .. -DCMAKE_INSTALL_PREFIX=your_install_path -DPYTHON_EXECUTABLE=$(which python)`, `your_install_path` must be in the root of DREAMPlace, even not in the subdirectory.

closed 2022-09-06 by SWang848