limbo018/DREAMPlace #86: GPU architecture mismatching

zerzerzerz, 2022-09-15:
- My GPU is RTX 2080 Ti and cuda is 11.0, when running `make` it raised `nvcc fatal : Unsupported gpu architecture 'compute_86'`
- I try to set `cmake -DCMAKE_CUDA_FLAGS=-gencode=arch=compute_60,code=sm_60` but it did not work and I found in cmake commands, it executed like `-gencode arch=compute_60,code=sm_60 -gencode arch=compute_61,code=sm_61 -gencode arch=compute_70,code=sm_70 -gencode arch=compute_75,code=sm_75 -gencode arch=compute_80,code=sm_80 -gencode arch=compute_86,code=sm_86`, so many same options in a line.
- Finally I modify `CMakeLists.txt` and delete the following lines `if (${CUDA_VERSION_MAJOR} VERSION_GREATER "10"); list(APPEND CMAKE_CUDA_ARCHITECTURES 8.0 8.6); endif(${CUDA_VERSION_MAJOR} VERSION_GREATER "10")`, it works
- Maybe it could be better to modify these GPU architecture settings
- Thanks for your excellent code for placement and routing!

limbo018, 2022-10-15:
Thanks for the suggestion! I searched around and found this is related to the CUDA version (nvcc version). As for as I know, cuda 11.0 should support compute_86. I haven't found a good solution to support most kinds of devices and CUDA versions yet, as GPU cards like RTX A40 and A6000 need compute_86. Will keep you updated if we solve this.

zerzerzerz, 2023-05-27:
After a year I meet this problem again~
My GPU device is RTX 3090 and CUDA is 11.6, it raised error `nvcc fatal : Unsupported gpu architecture 'compute_86'`.
Just remove `compute_86` in cmake file~

closed 2023-05-27 by zerzerzerz

limbo018, 2023-06-02:
Fixed at https://github.com/limbo018/DREAMPlace/commit/cefc7b87c3f2acb66d5f5be4b8c19ee51fe9b37d