limbo018/DREAMPlace #89: The value of dct seems wrong

BWbwchen, 2022-10-13:
Hi,

When I read the code of your [dct_unittest.py](https://github.com/limbo018/DREAMPlace/blob/master/unittest/ops/dct_unittest.py), It seems that your golden value isn't as the same as the other package's value.
In the [dct_unittest.py](https://github.com/limbo018/DREAMPlace/blob/master/unittest/ops/dct_unittest.py)'s [test_ct2Random function](https://github.com/limbo018/DREAMPlace/blob/4f80d9b30299c8b6ffa959b2d50006a52d8576ef/unittest/ops/dct_unittest.py#L170):

If I use **M = 4, N = 4** , the x will be : 
```
tensor([[4.5808, 4.8286, 3.1250, 6.1502],
        [2.1395, 4.1183, 6.9379, 9.6931],
        [6.1780, 3.3037, 5.4794, 4.4396],
        [7.0409, 5.5725, 6.9586, 9.8493]])
```
And your [discrete_spectral_transform.dct2_N](https://github.com/limbo018/DREAMPlace/blob/4f80d9b30299c8b6ffa959b2d50006a52d8576ef/dreamplace/ops/dct/discrete_spectral_transform.py#L82) will calculate the 2d dct value of x : 
```
[[22.598848   -2.8018274   1.7231263   0.10529643]
 [-2.146154   -0.30739462 -0.329887   -0.8330721 ]
 [ 1.0282258   0.59419966  0.5657164  -0.7704548 ]
 [-1.8328357   2.2623353   0.06582533  0.45616227]]
```
But, When I use [scipy.fftpack.dctn](https://docs.scipy.org/doc/scipy/reference/generated/scipy.fftpack.dctn.html) to calculate the 2d dct value of x, 
```python
import numpy as np
from scipy.fftpack import dctn
x2d = np.array([ 
        [4.5808, 4.8286, 3.1250, 6.1502],
        [2.1395, 4.1183, 6.9379, 9.6931],
        [6.1780, 3.3037, 5.4794, 4.4396],
        [7.0409, 5.5725, 6.9586, 9.8493]])
print(dctn(x2d, norm='ortho'))
```

I got : 
```
[[22.59885    -3.96235085  2.43685     0.1488571 ]
 [-3.03510712 -0.61479426 -0.65980756 -1.66615006]
 [ 1.4541      1.18833006  1.1314     -1.54089811]
 [-2.59202061  4.52469994  0.1316161   0.91234426]]
```

Only the [0][0] value is correct, and the other values are wrong.
I'm confused about this dct calculation, can you help me with this difference?

Thanks for your help.

limbo018, 2022-10-16:
Hi, it does not match the "ortho" mode.  Run ```print(dctn(x2d, norm=None) / 16)``` and you will get following results that match the golden results. 

```
[[22.59885   , -2.80180516,  1.72311316,  0.10525786],
       [-2.14614482, -0.30739713, -0.32990378, -0.83307503],
       [ 1.02820397,  0.59416503,  0.5657    , -0.77044906],
       [-1.83283535,  2.26234997,  0.06580805,  0.45617213]]
```

BWbwchen, 2022-10-23:
Hi,

Thanks for your help, it helps me a lot!

But your reply seems didn't work on the inverse dct case. When I test the [test_idct2Random function](https://github.com/limbo018/DREAMPlace/blob/4f80d9b30299c8b6ffa959b2d50006a52d8576ef/unittest/ops/dct_unittest.py#L250):

If I use **M = 2, N = 4** , the x will be : 
```
tensor([[4.5808, 4.8286, 3.1250, 6.1502],
        [2.1395, 4.1183, 6.9379, 9.6931]])
```
And the golden value of 2d idct value of x : 
```
[[18.323383 19.314262 12.499935 24.600866]
 [ 8.557888 16.473045 27.75176  38.772346]]
```
But, When I use [scipy.fftpack.idctn](https://docs.scipy.org/doc/scipy/reference/generated/scipy.fftpack.idctn.html) to calculate the 2d idct value of x, (without norm options)
```python
import numpy as np
from scipy.fftpack import idctn
x2d = np.array([[4.5808, 4.8286, 3.1250, 6.1502],
        [2.1395, 4.1183, 6.9379, 9.6931]])
print(idctn(x2d))
```

I got : 
```
[[ 60.78434837 -39.22881998  17.85140504  -8.98089377]
 [-15.52541617  24.21471049  -2.19176509  -0.27716889]]
```

As we can notice that there are some value are negative, while the golden value are all positive. 
It seems that we can not just divide a const number to solve mis-match problem.
I'm confused about this idct calculation, can you help me with this difference?

Thanks for your help.

BWbwchen, 2022-11-07:
@limbo018  Can you help me with this problem? Thank you very much!!

limbo018, 2022-11-07:
Hi, I checked your input and results on my side actually match the results of scipy. Did you misuse ```x``` and ```y```? 

The results of ```discrete_spectral_transform.idct2_2N(y).data.numpy()``` is as following: 
```
2D idct golden_value
[[ 60.784348   -39.22882     17.851408    -8.980894  ]
 [-15.525415    24.21471     -2.1917667   -0.27716875]]
```

BWbwchen, 2022-11-07:
> Did you misuse x and y?

Oh!! Yes, I misuse x and y...
Thanks for your help. I appreciate your help very much!

closed 2022-11-08 by BWbwchen