limbo018/DREAMPlace #94: I meet "_Map_base::at", when read def

LJK314, 2022-10-28:


YihangQiu, 2022-10-30:
Could you provide more detailed information ?

closed 2022-11-03 by LJK314

AlgorithmAI, 2026-11-02:
Traceback (most recent call last):
  File "/home/lichen/DACRUN/dreamplace/Placer.py", line 133, in <module>
    place(params)
  File "/home/lichen/DACRUN/dreamplace/Placer.py", line 34, in place
    placedb(params)
  File "/home/lichen/DACRUN/dreamplace/PlaceDB.py", line 478, in __call__
    self.read(params)
  File "/home/lichen/DACRUN/dreamplace/PlaceDB.py", line 421, in read
    self.rawdb = place_io.PlaceIOFunction.read(params)
  File "/home/lichen/DACRUN/dreamplace/ops/place_io/place_io.py", line 34, in read
    return place_io_cpp.forward(args.split(' '))
IndexError: map::at