limbo018/DREAMPlace #95: The parameter "target_density" in test/dac2012/*.json

YihangQiu, 2022-10-30:
Hi !
I found that the parameter "target_density" is equal to "0.9" in all dac2012/*.json.
[image]

But in DAC 2012 Benchmarksuit, it is recommended to set a specific value.
[image]

So, my question is, for the "target_density" parameter, do I keep the default parameter, or do I need to keep it consistent with the dac2012 dataset?  Specifically, for superblue19, I set this parameter to 0.78 instead of the default 0.9?

limbo018, 2022-10-30:
Hi, target density is a user-specific parameter and you can set it according to your needs. I checked old logs and notice that our previous results were collected with target_density = 0.9.

YihangQiu, 2022-10-30:
Thanks! 
I have another question, that is, I tested the DAC2012 dataset using the default parameters you provided in json, but the results came out to be quite different from the results in your paper. 

I don't quite understand where the problem might be. Thanks for your reply!

1. dac2012/superblue2.json
[image]
2. my results evaluated by dac2012_evaluate_solution.pl
[image]
3. your results
[image]

YihangQiu, 2022-10-30:
In addition, when I execute the program 2 times for the same input, there is randomness in the results. This randomness is not good for my algorithm comparison. **Could you tell me how I can eliminate this randomness?** Thank you for your reply!

limbo018, 2022-10-30:
Hi, I attached the old log for superblue2. One difference I notice is the detailed placement. We used ntuplace4h for detailed placement.

limbo018, 2022-10-30:
> In addition, when I execute the program 2 times for the same input, there is randomness in the results. This randomness is not good for my algorithm comparison. **Could you tell me how I can eliminate this randomness?** Thank you for your reply!

You can turn on the deterministic_flag in JSON, which can help alleviate the problem. The randomness comes from atomicAdd in CUDA, which can cause numerical indeterminism. If I remember correctly, the latest code used atomicAdd in the ops related to the electric force computation for efficiency, which may lose determinism. Setting ```deterministic_flag``` does not completely disable this in the latest release, which needs to be fixed later.

YihangQiu, 2022-10-30:
Thank you so much! 
Regarding the setting of num_bins, the result reported by the program does not match the json, which may be a bug. 
[image]

YihangQiu, 2022-10-30:
Thank you so much! I have no more questions and this issue can be closed.

closed 2022-10-30 by YihangQiu