limbo018/DREAMPlace #96: RUDY vs NCTUgr

YihangQiu, 2022-11-01:
Hi!
If I use RUDY instead of NCTUgr to get the route utilization map, the output only has the placement result but not the routing result. This way, I can't call the evaluation script of dac2012 to get the congestion metrics. So, my question is, if I use RUDY, how should I evaluate the congestion metrics?

The following are the output congestion metrics after calling the dac2012 evaluation script.
[image]

limbo018, 2022-11-01:
I'm confused about your question. Even using NCTUgr to get the route utilization map, the placer only generates .pl file as well, which is the expected behavior of a placer. I remember the evaluation script takes the .pl file as input and runs NCTUgr to get routability metrics. If you have problem with that, you can also refer to the congestion values reported by ```ntuplace4h```. It actually integrated NCTUgr for evaluation.

YihangQiu, 2022-11-03:
Hi! Prof. Lin. Thank you so much! 
Since calling NCTUgr keeps the intermediate routing result, I mistakenly took this result as the final routing result. 
Thanks for your answer, now I'm more clear about the whole congestion evaluation flow. And this issue can be closed.

closed 2022-11-03 by YihangQiu